jrowberg/i2cdevlib #135: MPU 6050 interrupts at 200Hz, but FIFO overflows to make it 160Hz

judepereira, 2015-01-16:
I tried benchmarking the frequency of the DMP example, and here's what I noticed:
- The MPU interrupts the code at 200Hz
- However, the reading of the data is limited to 160Hz
- The FIFO overflows, which is the reason for the drop in frequency

Here's the code that I'm using to do the benchmark (it's adapted from your example):
```
#include "I2Cdev.h"

#include "MPU6050_6Axis_MotionApps20.h"
#if I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE
    #include "Wire.h"
#endif

MPU6050 mpu;

uint32_t last = micros();
uint16_t i = 0;
volatile uint16_t ic = 0;
// MPU control/status vars
bool dmpReady = false;  // set true if DMP init was successful
uint8_t mpuIntStatus;   // holds actual interrupt status byte from MPU
uint8_t devStatus;      // return status after each device operation (0 = success, !0 = error)
uint16_t packetSize;    // expected DMP packet size (default is 42 bytes)
uint16_t fifoCount;     // count of all bytes currently in FIFO
uint8_t fifoBuffer[64]; // FIFO storage buffer

// packet structure for InvenSense teapot demo
uint8_t teapotPacket[14] = { '$', 0x02, 0,0, 0,0, 0,0, 0,0, 0x00, 0x00, '\r', '\n' };
volatile bool mpuInterrupt = false;     // indicates whether MPU interrupt pin has gone high
void dmpDataReady() {
    mpuInterrupt = true;
    ic++;
}
void setup() {
    #if I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE
        Wire.begin();
//        TWBR = 24; // 400kHz I2C clock (200kHz if CPU is 8MHz)
    #elif I2CDEV_IMPLEMENTATION == I2CDEV_BUILTIN_FASTWIRE
        Fastwire::setup(400, true);
    #endif
    Serial.begin(115200);
    mpu.initialize();

    // verify connection
    Serial.println(F("Testing device connections..."));
    Serial.println(mpu.testConnection() ? F("MPU6050 connection successful") : F("MPU6050 connection failed"));
    devStatus = mpu.dmpInitialize();
    if (devStatus == 0) {
        mpu.setDMPEnabled(true);
        attachInterrupt(23, dmpDataReady, RISING);
        mpuIntStatus = mpu.getIntStatus();
        dmpReady = true;
        packetSize = mpu.dmpGetFIFOPacketSize();
    } else {
        Serial.print(F("DMP Initialization failed (code "));
        Serial.print(devStatus);
        Serial.println(F(")"));
    }
}
void loop() {
    if (!dmpReady) return;
    mpuInterrupt = false;
    mpuIntStatus = mpu.getIntStatus();
    fifoCount = mpu.getFIFOCount();
    if ((mpuIntStatus & 0x10) || fifoCount == 1024) {
        mpu.resetFIFO();
        Serial.println(F("FIFO overflow!"));
    } else if (mpuIntStatus & 0x02) {
        while (fifoCount < packetSize) fifoCount = mpu.getFIFOCount();
        mpu.getFIFOBytes(fifoBuffer, packetSize);
        fifoCount -= packetSize;
        i++;
        if(micros() -last > 1000000) {
            Serial.println(i);
            Serial.println(ic);
            last = micros();
            i=0;
            ic = 0;
        }
    }
}
```

BTW, great work on the lib!

judepereira, 2015-01-16:
Also note, I'm on the Arduino Due, so the processor is pretty fast.

judepereira, 2015-01-24:
Okay, changing the i2c bus clock speed in variant.cpp worked for me :)

closed 2015-01-24 by judepereira

traveltrousers, 2015-05-24:
You mention the variant.cpp twice... where is it?

judepereira, 2015-06-18:
/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/sam/variants/arduino_due_x/variant.cpp

duecedriver, 2015-12-18:
anyone smart on this possibly jump over to the Stratux project and see if you can give us a hand implementing the RY835AI on the PI

it uses the 6050 but reyax has terrible docs, we dont know if we are getting raw angles or fused DMP angles since in flight the unit does not provide a good horizon.

it could also be the way folks are hooking into the i2c... 
thanks in advance

ohsm88, 2015-12-18:
NAVER - http://www.naver.com/
--------------------------------------------

ohsm88@naver.com 님께 보내신 메일 <Re: [i2cdevlib] MPU 6050 interrupts at 200Hz, but FIFO overflows to make it 160Hz (#135)> 이 다음과 같은 이유로 전송 실패했습니다.

--------------------------------------------

받는 사람이 회원님의 메일을 수신차단 하였습니다. 

--------------------------------------------