jrowberg/i2cdevlib #137: dmpGetMag returning zeros

marcbishara, 2015-02-03:
using function:
https://github.com/marcbishara/i2cdevlib/blob/master/Arduino/MPU6050/MPU6050_9Axis_MotionApps41.h#L753

Returns 0,0,0 Is that an outstanding issue or am I doing something wrong in setup?

Monax111, 2016-02-28:
I have the same problem!

Monax111, 2016-02-28:
I have the Arduino Nano, 2 MPU6050, 2 HMC5883L
I want to get the output array with 20 values (the first board {3, 3, accelerometer, magnetometer 3, temperature}), the second board {3, 3, accelerometer, magnetometer 3, temperature})
Now I'm trying to make it work the first board (MPU6050+HMC5883L series connection).
But i have the magnetometer (0.00,0.00,0.00);
What do you think, what's the problem?
[my scheme](https://pp.vk.me/c630530/v630530113/1b27d/Q8F3Sna7UU8.jpg)

Recontech, 2016-02-29:
Have you tried adding pull-up resistors?

yang614, 2016-07-07:
The AK8975's address is 0x0C,but the code give the address is 0x0E, so change the address to 0x0C.