jrowberg/i2cdevlib #142: Deprecated type in avr-libc 1.8.0

kerny3d, 2015-03-05:
The Arduino version of the i2cdevlib can not be built on avr-lib 1.8.0+ (on Arduino IDE 1.0.5). The problem is caused by deprecated data types in that version of avr-lib. An example log of Arduino IDE is the following:

In file included from test.ino:20:0:
Arduino/libraries/MPU9150/MPU9150_9Axis_MotionApps41.h:124:7: error: 'prog_uchar' does not name a type
 const prog_uchar dmpMemory[MPU9150_DMP_CODE_SIZE] PROGMEM = {
       ^
Arduino/libraries/MPU9150/MPU9150_9Axis_MotionApps41.h:265:7: error: 'prog_uchar' does not name a type
 const prog_uchar dmpConfig[MPU9150_DMP_CONFIG_SIZE] PROGMEM = {
       ^
Arduino/libraries/MPU9150/MPU9150_9Axis_MotionApps41.h:315:7: error: 'prog_uchar' does not name a type
 const prog_uchar dmpUpdates[MPU9150_DMP_UPDATES_SIZE] PROGMEM = {

As written on:

http://www.nongnu.org/avr-libc/user-manual/group__avr__pgmspace.html#gaa475b6b81fd8b34de45695da1da523b6

 A workaround can be to define the macro PROG_TYPES_COMPAT before the inclusion of <avr/pgmspace.h> but i think this can be a problem in Arduino IDE.

A possible solution can be to change types of problematic variables in the original source code of the library.

jrowberg, 2015-04-16:
This has just been solved in a merged pull request from @Bonobo79. Thanks!

closed 2015-04-16 by jrowberg

Boobies, 2015-04-20:
Are you sure this has been fixed? I'm getting the same error with the latest revision (#159) and Arduino 1.6.3 for Windows.

Cheers,
Bogdan

Boobies, 2015-04-27:
If I merge the code for the two MPU's, will you accept it? I want to know if it's worth wasting my time with or if I should just fix the MPU9150 code for my particular project.