jrowberg/i2cdevlib #143: Two MPU6050 sensors to work with Arduino Uno

bgw2117, 2015-03-07:
I am trying to read yaw, pitch and roll from two mpu6050 sensor simultaneously using Arduino Uno. The sensors are from sparkfun. I got A0 as high in one chip and low in the other (0x69 and 0x68).

I set up the sensors parallel as in:
A5 - SCL_1 - SCL_2
A4 - SDA_1 - SDA_2

3.3V - VIO_1 -- VIO_2 -- VDD_1 - VDD_2
GND -- GND_1 - GND_2

I also created a new interrupt for the second sensor in Jeff's code. 

INT_1 -- D2
INT_2 -- D3

After all of this, I am still getting inconsistent results. Anyone has any idea why that might be?

sinisterdev, 2015-03-20:
use a separate object for the new sensor
MPU6050 mpu;
MPU6050 mpu1(0X69);

Tirocinio, 2015-08-06:
Hello everybody,
We should connect two mpu6050 to an Arduino mega but we don't know how to connect them physically.
We know that we have to turn the i2c adress from 0x68 to 0x69 in order to have 2 different adress, one for each sensor.
Someone says that it must connect them in parallel but other thinks that it is usefull to set two digital pins high or low to "swich on/off" a sensor in order to read information alternated.
moreover we would like to read quarternion or euler data but now we are only able to receive raw data.
May anyone help us?
thanks

sinisterdev, 2015-08-06:
@Tirocinio connect it in parallel physically, SDA SCL to sensor 1 and same to sensor 2 also. Make sure you change the address of the second sensor by connecting the address pin to high or soldering the AD0 patch(Depending on break out board). And switching sensors alternately beats the purpose of using the I2C interface.