jrowberg/i2cdevlib #145: How Can I get Gyro data using the MPU6050_test library

ozanio, 2015-03-14:
I have tried adding the line mpu.dmpGetGyro(gyro, fifoBuffer); as below

mpu.dmpGetQuaternion(&q, fifoBuffer);
mpu.dmpGetGyro(gyro, fifoBuffer);
mpu.dmpGetGravity(&gravity, &q);
mpu.dmpGetYawPitchRoll(ypr, &q, &gravity);

but I get a "gyro was not declared in this scope error" so I added the line "float gyro[3];" at the top of the code where float euler[3]; was which gave me a "no matching function error" 

How can I fix this?

findftp, 2015-03-15:
I guess you should post the complete code, makes it easier for others to debug.

Arrowana, 2015-07-08:
use getMotion6()

mpu.getMotion6(&ax, &ay, &az, &gx, &gy, &gz);

to get all 3 accelerations and gyroscopes

sguillia, 2017-03-09:
Use `int gyro[3]` instead of `float gyro[3]`