jrowberg/i2cdevlib #146: uint8_t getMotionStatus(); should be    bool getMotionStatus();

findftp, 2015-03-15:
Hello,

This is my first contribution on github.
While compiling code on my arduino, I got errors.
Even the example code spits out errors like this:

/home/michiel/sketchbook/libraries/MPU6050/MPU6050.cpp:1997: error: prototype for ‘bool MPU6050::getMotionStatus()’ does not match any in class ‘MPU6050’
/home/michiel/sketchbook/libraries/MPU6050/MPU6050.h:611: error: candidate is: uint8_t MPU6050::getMotionStatus()

So, I went to MPU6050.h and changed the uint8_t getMotionStatus(); into bool getMotionStatus();

Now the sketch compiles without error and spits out values so it seems to be working now.
I'm not sure what getMotionStatus(); does, I'm a noob coder but I wanted to let you know this problem so that more intelligent people can check if I fixed the problem like it should.

snalesso, 2015-03-18:
I can confirm, same problem, same solution.

collirob5, 2015-03-20:
Confirmed again.

bernji, 2015-03-22:
I can at lease confirm that the example code is not compiling with the error:
C:\Users\XXX\Documents\Arduino\libraries\MPU6050\MPU6050.h:611:17: error: candidate is: uint8_t MPU6050::getMotionStatus()
         uint8_t getMotionStatus();
                 ^

projectnew001, 2015-03-23:
Confirmed!

pacojdc, 2015-03-29:
opened an closed a new issue for this same problem. I confirm also the issue.

eadf, 2016-04-16:
This seems to be fixed now. 
MPU6050.h : uint8_t getMotionStatus();
MPU6050.cpp : uint8_t MPU6050::getMotionStatus() {