jrowberg/i2cdevlib #153: modifying the arduino code for second MPU-6050 (How to expand code for second MPU-6050)

ferizitko, 2015-03-29:
Hi there,
I'm currently working on a project related to MPU-6050, arduino, and labview software for simulation. I have poor knowledge in C or C++ PROGRAMMING. I need to use 2 mpu-6050 sensors to capture the movement of human arm. I have downloaded one library for both arduino and labview and it is working great. the arduino library retrieve the raw values from MPU-6050, then convert them to meaningful data (3 axes accel, 3 axes gyroscope), and finally it filters the data by both complementary and Kalman filters. the result is awesome but......
the problem: I need to add another mpu-6050 to arduino, I know how the connection will be for the second sensor and i know how to use the address 68 for one and 69 for second imu. but the problem is i dont know which section(s) of program i should modify.
Note: the arduino library that I attach to this post looks huge but i believe that if anyone has experience with programming can easily understand which sections must be modified.
I'm completely confused while i was checking this library.
to those who recommend to use i2cdev library: i have used this library and it was awesome but i need to filter the data by either Klaman and complementy as well and that is really hard to implement in labview for me.

// MPU-6050 Accelerometer + Gyro + Arduino Uno
// -----------------------------
//
// 
// June 2012
// Open Source / Public Domain
//
// Using Arduino 1.0.1
// It will not work with an older version, 
// since Wire.endTransmission() uses a parameter 
// to hold or release the I2C bus.
//
// Documentation:
// - The InvenSense documents:
//   - "MPU-6000 and MPU-6050 Product Specification",
//     PS-MPU-6000A.pdf
//   - "MPU-6000 and MPU-6050 Register Map and Descriptions",
//     RM-MPU-6000A.pdf or RS-MPU-6000A.pdf
//   - "MPU-6000/MPU-6050 9-Axis Evaluation Board User Guide"
//     AN-MPU-6000EVB.pdf
// 
// The accuracy is 16-bits.
//
// Temperature sensor from -40 to +85 degrees Celsius
//   340 per degrees, -512 at 35 degrees.
//
// At power-up, all registers are zero, except these two:
//      Register 0x6B (PWR_MGMT_2) = 0x40  (I read zero).
//      Register 0x75 (WHO_AM_I)   = 0x68.
// 

#include <Wire.h>

// Register names according to the datasheet.
// According to the InvenSense document 
// "MPU-6000 and MPU-6050 Register Map 
// and Descriptions Revision 3.2", there are no registers
// at 0x02 ... 0x18, but according other information 
// the registers in that unknown area are for gain 
// and offsets.
// 
#define MPU6050_AUX_VDDIO          0x01   // R/W
#define MPU6050_SMPLRT_DIV         0x19   // R/W
#define MPU6050_CONFIG             0x1A   // R/W
#define MPU6050_GYRO_CONFIG        0x1B   // R/W
#define MPU6050_ACCEL_CONFIG       0x1C   // R/W
#define MPU6050_FF_THR             0x1D   // R/W
#define MPU6050_FF_DUR             0x1E   // R/W
#define MPU6050_MOT_THR            0x1F   // R/W
#define MPU6050_MOT_DUR            0x20   // R/W
#define MPU6050_ZRMOT_THR          0x21   // R/W
#define MPU6050_ZRMOT_DUR          0x22   // R/W
#define MPU6050_FIFO_EN            0x23   // R/W
#define MPU6050_I2C_MST_CTRL       0x24   // R/W
#define MPU6050_I2C_SLV0_ADDR      0x25   // R/W
#define MPU6050_I2C_SLV0_REG       0x26   // R/W
#define MPU6050_I2C_SLV0_CTRL      0x27   // R/W
#define MPU6050_I2C_SLV1_ADDR      0x28   // R/W
#define MPU6050_I2C_SLV1_REG       0x29   // R/W
#define MPU6050_I2C_SLV1_CTRL      0x2A   // R/W
#define MPU6050_I2C_SLV2_ADDR      0x2B   // R/W
#define MPU6050_I2C_SLV2_REG       0x2C   // R/W
#define MPU6050_I2C_SLV2_CTRL      0x2D   // R/W
[... 1081 lines cut ...]
//This function pads the printed value to take up 
//7 characters Example: **20.16
void pad(float PadVal){
  
  //Positve Value
  if (PadVal > 0){          //For positive value
    if(PadVal < 10){        //For positive value less than 10
     Serial.print("***");   //0-9 ***0-9.00
 }  
 else if (PadVal < 100){    //values less than 100
 Serial.print("**");        //10-99 **10-99.00
 }
    
  }
  else if(PadVal < 0){      //for negative values
    if(PadVal > -10){        //for negative value greater than -10 
     Serial.print("**");     //-9 - 0 **-1 - -9.00
 }  
 else if (PadVal > -100){   //for negative values lesss than -99
 Serial.print("*");          //*-99.00
 }
  }
    
 Serial.print(PadVal, 2);  //print the padded value
}

Tirocinio, 2015-07-29:
Hi @ferizitko , 
we have some questions for you:
first of all have you solved your problem, how?
We should connect two mpu6050 to an Arduino mega but we don't know how to connect them physically. 
Someone says that we must connect them in parallel but other thinks that it is usefull to set two digital pins high or low to "swich on/off" a sensor in order to read information coming from the other.
We would like to read quarternion or euler data but now we are only able to receive raw data.
may you help us?
thanks

ranqingfa, 2015-07-30:
Hi @ferizitko ,
Have you took notice of the code below ?

// class default I2C address is 0x68
// specific I2C addresses may be passed as a parameter here
// AD0 low = 0x68 (default for InvenSense evaluation board)
// AD0 high = 0x69
MPU6050 accelgyro;
//MPU6050 accelgyro(0x69); // <-- use for AD0 high

Sanjay44NS, 2018-01-30:
hey, @ferizitko 
We are also working on a similar project.
Did u get any positive results for two IMUs??
if so help us!!
thank you.