jrowberg/i2cdevlib #154: MPU6050_DMP6 example crashing

opvini, 2015-03-29:
Hi everyone.
I am using an Arduino Mega, and the MPU6050. I tested the MPU6050_DMP6 example and the Arduino crashing after a while. after extensive testing, I am reading raw data directly with getMotion6().

I posted on http://forum.arduino.cc/index.php?topic=166938.msg2163176#msg2163176

Some one had the same problem?
Thanks a lot!

opvini, 2015-03-29:
I changed the follow line on MPU6050_DMP6 example and it works now!!!

 TWBR = 24;
to
 TWBR = 12;

opvini, 2015-03-29:
The line 
 TWBR = 24;
is correct to my Arduino Mega with 16MHz?

opvini, 2015-03-29:
Here is the code, using 2 servos and without serial.

[code]

#include "I2Cdev.h"
#include "MPU6050_6Axis_MotionApps20.h"

#if I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE
    #include "Wire.h"
#endif

MPU6050 mpu;

#include <Servo.h>

Servo myservo; 
Servo myservo2; 
int valAtual = 0;
int valAtual2 = 0;
int lastVal  = 0;

#define LED_PIN 13 // (Arduino is 13, Teensy is 11, Teensy++ is 6)
bool blinkState = false;

// MPU control/status vars
bool dmpReady = false;  // set true if DMP init was successful
uint8_t mpuIntStatus;   // holds actual interrupt status byte from MPU
uint8_t devStatus;      // return status after each device operation (0 = success, !0 = error)
uint16_t packetSize;    // expected DMP packet size (default is 42 bytes)
uint16_t fifoCount;     // count of all bytes currently in FIFO
uint8_t fifoBuffer[64]; // FIFO storage buffer

// orientation/motion vars
Quaternion q;           // [w, x, y, z]         quaternion container
VectorInt16 aa;         // [x, y, z]            accel sensor measurements
VectorInt16 aaReal;     // [x, y, z]            gravity-free accel sensor measurements
VectorInt16 aaWorld;    // [x, y, z]            world-frame accel sensor measurements
VectorFloat gravity;    // [x, y, z]            gravity vector
float euler[3];         // [psi, theta, phi]    Euler angle container
float ypr[3];           // [yaw, pitch, roll]   yaw/pitch/roll container and gravity vector

// packet structure for InvenSense teapot demo
uint8_t teapotPacket[14] = { '$', 0x02, 0,0, 0,0, 0,0, 0,0, 0x00, 0x00, '\r', '\n' };

// ================================================================
// ===               INTERRUPT DETECTION ROUTINE                ===
// ================================================================

volatile bool mpuInterrupt = false;     // indicates whether MPU interrupt pin has gone high
void dmpDataReady() {
    mpuInterrupt = true;
}

// ================================================================
// ===                      INITIAL SETUP                       ===
// ================================================================

void setup() {    
    
    // join I2C bus (I2Cdev library doesn't do this automatically)
    #if I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE
        Wire.begin();
        // antes era 24
        TWBR = 12; // 400kHz I2C clock (200kHz if CPU is 8MHz)
    #elif I2CDEV_IMPLEMENTATION == I2CDEV_BUILTIN_FASTWIRE
        Fastwire::setup(400, true);
    #endif
    
    // initialize device
    mpu.initialize();
[... 81 lines cut ...]
            mpu.dmpGetQuaternion(&q, fifoBuffer);
            mpu.dmpGetGravity(&gravity, &q);
            mpu.dmpGetYawPitchRoll(ypr, &q, &gravity);
            
            /*
            Serial.print("ypr\t");
            Serial.print(ypr[0] * 180/M_PI);
            Serial.print("\t");
            Serial.print(ypr[1] * 180/M_PI);
            Serial.print("\t");
            Serial.println(ypr[2] * 180/M_PI);
             */           
            
        valAtual  = 90 - (ypr[1] * 180/M_PI);          
        valAtual2 = 90 - (ypr[2] * 180/M_PI);
   
        myservo.write(valAtual);
        myservo2.write(valAtual2); 
        
        // blink LED to indicate activity
        blinkState = !blinkState;
        digitalWrite(LED_PIN, blinkState);
    }
}
[/code]

wolframalpha, 2015-07-09:
Same problem here ... I guess the problem is with the buffer since sometimes it ain't able to read the register ! Did you get a solution to it ??

aborjesson, 2015-10-10:
I have the same problem. After a while reading the registers, either FIFO Count or Raw acc/gyro values, makes it crash. Any new findings to that problem?

wolframalpha, 2015-10-11:
@aborjesson Okay ... There was a problem reading the on-board registers in I2Cdev.cpp,which was annoying and frustrating,huh!! Anyways Fixed it , here's the link https://drive.google.com/file/d/0BwUusXozWAMjVWhMYkZ4NlZwYlk/view?usp=sharing to the modified library. Override the original file and you are good to go. Ping me if it works.

aborjesson, 2015-10-11:
@wolframalpha: I diffed your I2CDev.cpp with the latest verison from @jrowberg and there was no code changes. So it did'nt sort out my mystery.

I think my problem is related to that I use the FIFO for DMP values for Yaw, Pitch, Roll and at the same time (while FIFO is getting filled from DMP 100Hz) read raw values from the registers for gyro and accel. Do anyone know if this is doable, to use the DMP/FIFO and at the same time read raw values from the registers ? Cannot find anything related to that in the data sheet.

opvini, 2015-11-06:
COOL!!! Now it is working fine!!!! Congrats.

opvini, 2015-11-08:
If I use a Servo, the program is crashing yet... I dont know if it is interference...
I am use a simple raw reading example. With no servos, it works fine.
What do you think?

ranqingfa, 2015-11-09:
Maybe the power supply is not enough！

opvini, 2015-11-09:
@ranqingfa I think can be a bug on wire arduino library. I am using a PC power supply (250W) and just 2 servos.

aborjesson, 2015-11-09:
@opvini : Those MPU-9150s are mysterious. Seem to be very sensitive to noise of all sort. I bought a new one, resoldered my whole sketch onto a pcb hole bread board. Now it seems to work rock solid even if I use the DMP. I do not use interrupts though, havent had the time to check that.

It is powered by a separate power supply, soldered onto the board.

opvini, 2015-11-10:
Hi guys! I solved the problem:

1) I changed the pull-ups resistors to 2k2
2) I put a ceramic capacitor between VCC and GND (104)
3) The main thing for me: I put a Ferrite Bead on my PWM line

Now it works very well and not freezing :D
Thanks a lot for everybody!

opvini, 2015-11-18:
Other important consideration: when I use some delay's on my firmware, it's crashes. Then I am using micros() to count time and execute the correct actions on a correct time.

109JB, 2015-11-27:
I am just learning arduino and C/C++ programming so I don't know if this will help, but I also found that the MPU6050 DMP demo program would crash.  I was able to get it to work reliably by eliminating the interrupt detection for the FIFO buffer and instead just poll the MPU6050 for the buffer state. Doing this eliminated the problems.I was having. 

Here is a cut and paste of my modified Arduino code. I noted where I commented out portions of the code.

// I2C device class (I2Cdev) demonstration Arduino sketch for MPU6050 class using DMP (MotionApps v2.0)
// 6/21/2012 by Jeff Rowberg <jeff@rowberg.net>
// Updates should (hopefully) always be available at https://github.com/jrowberg/i2cdevlib
//
// Changelog:
//      2013-05-08 - added seamless Fastwire support
//                 - added note about gyro calibration
//      2012-06-21 - added note about Arduino 1.0.1 + Leonardo compatibility error
//      2012-06-20 - improved FIFO overflow handling and simplified read process
//      2012-06-19 - completely rearranged DMP initialization code and simplification
//      2012-06-13 - pull gyro and accel data from FIFO packet instead of reading directly
//      2012-06-09 - fix broken FIFO read sequence and change interrupt detection to RISING
//      2012-06-05 - add gravity-compensated initial reference frame acceleration output
//                 - add 3D math helper file to DMP6 example sketch
//                 - add Euler output and Yaw/Pitch/Roll output formats
//      2012-06-04 - remove accel offset clearing for better results (thanks Sungon Lee)
//      2012-06-01 - fixed gyro sensitivity to be 2000 deg/sec instead of 250
//      2012-05-30 - basic DMP initialization working

/* ============================================
I2Cdev device library code is placed under the MIT license
Copyright (c) 2012 Jeff Rowberg

Permission is hereby granted, free of charge, to any person obtaining a copy
of this software and associated documentation files (the "Software"), to deal
in the Software without restriction, including without limitation the rights
to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
copies of the Software, and to permit persons to whom the Software is
furnished to do so, subject to the following conditions:

The above copyright notice and this permission notice shall be included in
all copies or substantial portions of the Software.

THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
THE SOFTWARE.
===============================================
*/

// I2Cdev and MPU6050 must be installed as libraries, or else the .cpp/.h files
// for both classes must be in the include path of your project
#include "I2Cdev.h"

#include "MPU6050_6Axis_MotionApps20.h"
//#include "MPU6050.h" // not necessary if using MotionApps include file

// Arduino Wire library is required if I2Cdev I2CDEV_ARDUINO_WIRE implementation
// is used in I2Cdev.h
#if I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE
    #include "Wire.h"
#endif

// class default I2C address is 0x68
// specific I2C addresses may be passed as a parameter here
// AD0 low = 0x68 (default for SparkFun breakout and InvenSense evaluation board)
// AD0 high = 0x69
MPU6050 mpu;
//MPU6050 mpu(0x69); // <-- use for AD0 high

/* =========================================================================
   NOTE: In addition to connection 3.3v, GND, SDA, and SCL, this sketch
   depends on the MPU-6050's INT pin being connected to the Arduino's
[... 276 lines cut ...]
            Serial.print("\t");
            Serial.print(aaWorld.y);
            Serial.print("\t");
            Serial.println(aaWorld.z);
        #endif
    
        #ifdef OUTPUT_TEAPOT
            // display quaternion values in InvenSense Teapot demo format:
            teapotPacket[2] = fifoBuffer[0];
            teapotPacket[3] = fifoBuffer[1];
            teapotPacket[4] = fifoBuffer[4];
            teapotPacket[5] = fifoBuffer[5];
            teapotPacket[6] = fifoBuffer[8];
            teapotPacket[7] = fifoBuffer[9];
            teapotPacket[8] = fifoBuffer[12];
            teapotPacket[9] = fifoBuffer[13];
            Serial.write(teapotPacket, 14);
            teapotPacket[11]++; // packetCount, loops at 0xFF on purpose
        #endif

        // blink LED to indicate activity
        blinkState = !blinkState;
        digitalWrite(LED_PIN, blinkState);
    }
}

109JB, 2015-11-27:
I forgot to mention that I alsi had to unhook the interrupt wire to get it to work properly.

jaundice, 2015-11-27:
I didn't have issues with the 6050 but had trouble with fifo overruns on 9150. Switching from wire to fastwire fixed it for me.

// 6/21/2012 by Jeff Rowberg jeff@rowberg.net<mailto:jeff@rowberg.net>

http://arduino.cc/forum/index.php/topic,109987.0.html
http://code.google.com/p/arduino/issues/detail?id=958

  *   ========================================================================= */

// http://en.wikipedia.org/wiki/Gimbal_lock)

opvini, 2015-12-01:
@jrowberg @wolframalpha After I updated the library, the DMP mode is working fine, but now, the raw values example firmware is crashing o.O

prvnkmr96, 2018-03-10:
What is the use of INT pin on MPU6050. Is it give interrupt only when the MPU in the motion or something else.

IshankAgarwal, 2019-02-23:
I am also using MPU6050 with Arduino Mega and motors with motor driver-Cytron MDDS-10A.
whenever I turn motors ON , Arduino Serial freezes up. But without motors connected  Mpu6050 works correct . If this is due to interference from motors ,
@opvini Then approx. what parameters of ferrite bead and ceramic capacitor should I use to decrease the noise from motors .Motors I am using are 12V DC 350rpm encoder motors (IG42E).

I tried with internal and external pullups of 2k2 both but still the same problem .