jrowberg/i2cdevlib #164: MPU6050 dmpInitialize hangs in Fastwire mode

nfdj, 2015-05-02:
Hello,
I found a bug in the MPU6050 library.
When using I2CDEV_IMPLEMENTATION=I2CDEV_BUILTIN_FASTWIRE, the dmpInialize functions hangs here:
...
DEBUG_PRINTLN(F("Waiting for FIFO count > 2..."));
            while ((fifoCount = getFIFOCount()) < 3);
...
(in file MPU6050_6Axis_MotionApps20.h)
 
Could someone guide me how to solve this bug?

antonmeyer, 2015-05-16:
Hi,
I observed similar (I did not break it down to the root cause in the software). In my case it was related to the 3.3V and 5V. Strange enough, running the logic level on 5 V works, where 3.3 V does not. (I use a seeduino) But that only happens when I run it on a accu which has 5.2 V where the power plug has 5.1 V.

So might be your problem has as root cause the HW but the SW does not react proper on this.

berndpfrommer, 2015-08-14:
I have the same issue: it hangs in dmpInitialize().
Tried two different IMUs, with and without 3.3/5V level shifters.