jrowberg/i2cdevlib #169: Newbie directory related problem (MPU6050)

Ryeg, 2015-05-26:
I'm 73 years old and trying to keep sharp by playing with interesting things, but I have a problem – I am working with I2Cdevlib and have some things from the package running, but am having a problem with others that seem to be include related (probably library related).

I am kind of a newbie with Arduino but used to pretty good with PIC C stuff.  I apologize if this is somewhere in the forum, but I have done my best to find something related without much luck.  If somebody could take a look at this and steer me in the right direction, I would be very grateful.  

This is probably my ineptitude with Arduino but..... I will try to describe the situation.

I am using windows 7 and the current Jeff Grawburg I2Cdevlib package.  

I can load the zip into my Arduino>library folder using the import library function of the Arduino IDE and can load and verify some examples successfully. Specifically,  (from I2Cdevlib library folder) MyDocuments >Arduino>libraries> I2Cdevlib>Examples>Arduino>ITG3200>Examples>ITG3200_raw  without any errors.

However, when I try to load and verify a similar file for a different device,   Arduino>libraries> I2Cdevlib-master >Arduino> MPU6050> Examples> MPU6050_raw,  I get the following errors that are making me crazy.

____________________________________________
Using library I2Cdev in folder: C:\Users\Rye\Documents\Arduino\libraries\I2Cdev (legacy)
Using library Wire in folder: C:\Program Files\Arduino\hardware\arduino\avr\libraries\Wire 

C:\Program Files\Arduino/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10703 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -IC:\Program Files\Arduino\hardware\arduino\avr\cores\arduino -IC:\Program Files\Arduino\hardware\arduino\avr\variants\mega -IC:\Users\Rye\Documents\Arduino\libraries\I2Cdev -IC:\Program Files\Arduino\hardware\arduino\avr\libraries\Wire C:\Users\Rye\AppData\Local\Temp\build2904214137857257350.tmp\MPU6050_raw.cpp -o C:\Users\Rye\AppData\Local\Temp\build2904214137857257350.tmp\MPU6050_raw.cpp.o 
MPU6050_raw.ino:37:21: fatal error: MPU6050.h: No such file or directory
compilation terminated.
Error compiling.
______________________________________

MPU6050.h is right there in the library directory…..

Being a newbie, I am stuck and thought somebody might be able to help or forward me to somebody who might be kind enough to help me.

At age 72 I am starting to doubt my skills (and sanity).

Thanks in advance for any help you might be able to give me.

Regards
Rye

smallzzy, 2015-06-30:
Well. In your case(I believe MPU6050.h is under \libraries\I2Cdev\Arduino\MPU6050), arduino is not able to find the file because the compiler does not search into the library directory (which is normal for most compilers)

So the solution will simply be: move all directory under \libraries\I2Cdev\Arduino to \libraries
in this way, all libraries for arduino are properly installed and the compiler is able to find them.

I hope my answer helps you. Technically, this is not a bug and it should not be reported here. A better place for problem like this would be the official forum.

gedw, 2015-10-29:
Hi Ryeg,
Welcome to the world of fussy compilers, may your sanity be strengthed by this AVR.
Now, I'm not really a noob,but I am medically retired (courtesy of a few strokes), so I 
understand the drive to keep the grey matter firing. When I was a younger programmer, I
was always chastened by the the more senior folks to always look closely at the error
listing, because the answer is usually in there.
I note that you changed target devices between compiles;
Did you also change the device via the IDE? Your error listing says that it's compiling for
an 'atmega2560' and my guess is that you were (before the device change) compiling for
a smaller chip (maybe a UNO with a 328p). I find that there's nothing to be gained by 
compiling for for boards you don't have yet, except hours of frustration leading to head pain.
Hope this helps,
Cheers,
ged
PS: Reality check are you 73 or 72 ? I'm 61 :-)

wagihr, 2018-10-02:
"MPU6050_6Axis_MotionApps20.h" no such file or directory
i need to solve it