jrowberg/i2cdevlib #183: Compile MPU6050 for Arduino Mega with Eclipse

ne0h, 2015-08-08:
Hi,

I am trying to compile the MPU6050 library with eclipse. I am getting many errors (about 300) like:

`../src/MPU6050.cpp: In member function 'bool MPU6050::getZeroMotionDetected()':
../src/MPU6050.cpp:2119:96: error: invalid conversion from 'uint8_t* {aka unsigned char*}' to 'int8_t* {aka signed char*}' [-fpermissive]
     I2Cdev::readBit(devAddr, MPU6050_RA_MOT_DETECT_STATUS, MPU6050_MOTION_MOT_ZRMOT_BIT, buffer);`

Do you have any idea what could be wrong?

Thanks in advance!
Max

eadf, 2016-04-18:
I tried the latest 'Arduino Eclipse Extensions 3.0' and could not see any errors.
The error you got indicates a signed/unsigned mismatch in the method declarations vs definitions.
I could not find that problem in the current code, it's probably fixed.

Could you test again?

i3130002, 2019-07-30:
It seems that i'm having the same issue using Arduino IDE 1.8.9 and VScode official arduino extension.  What do you suggest?  Also I wasn't able to find 'Arduino Eclipse Extensions 3.0', is it the same as [This ](https://www.hackster.io/dallas-onem2m/using-eclipse-arduino-plug-in-with-esp8266-nodemcu-board-cb77ef) or [This](http://eclipse.baeyens.it/stable.php) please give us a link to the IDE and settings you are using.

[image]

```
sketch\MPU6050.cpp: In member function 'void MPU6050::PID(uint8_t, float, float, uint8_t)':

MPU6050.cpp:3258:65: error: invalid conversion from 'int16_t* {aka short int*}' to 'uint16_t* {aka short unsigned int*}' [-fpermissive]

   I2Cdev::readWords(devAddr, SaveAddress + (i * shift), 1, &Data); // reads 1 or more 16 bit integers (Word)

                                                                 ^

In file included from sketch\MPU6050.h:40:0,

                 from sketch\MPU6050.cpp:38:

I2Cdev.h:115:23: error: initializing argument 4 of 'static int8_t I2Cdev::readWords(uint8_t, uint8_t, uint8_t, uint16_t*, uint16_t)' [-fpermissive]

         static int8_t readWords(uint8_t devAddr, uint8_t regAddr, uint8_t length, uint16_t *data, uint16_t timeout=I2Cdev::readTimeout);

                       ^

MPU6050.cpp:3272:63: error: invalid conversion from 'int16_t* {aka short int*}' to 'uint16_t* {aka short unsigned int*}' [-fpermissive]

     I2Cdev::readWords(devAddr, ReadAddress + (i * 2), 1, &Data); // reads 1 or more 16 bit integers (Word)

                                                               ^

In file included from sketch\MPU6050.h:40:0,

                 from sketch\MPU6050.cpp:38:

I2Cdev.h:115:23: error: initializing argument 4 of 'static int8_t I2Cdev::readWords(uint8_t, uint8_t, uint8_t, uint16_t*, uint16_t)' [-fpermissive]

         static int8_t readWords(uint8_t devAddr, uint8_t regAddr, uint8_t length, uint16_t *data, uint16_t timeout=I2Cdev::readTimeout);

                       ^

MPU6050.cpp:3283:69: error: invalid conversion from 'int16_t* {aka short int*}' to 'uint16_t* {aka short unsigned int*}' [-fpermissive]

     I2Cdev::writeWords(devAddr, SaveAddress + (i * shift), 1,  &Data);

                                                                     ^

In file included from sketch\MPU6050.h:40:0,

                 from sketch\MPU6050.cpp:38:

I2Cdev.h:124:21: error: initializing argument 4 of 'static bool I2Cdev::writeWords(uint8_t, uint8_t, uint8_t, uint16_t*)' [-fpermissive]

         static bool writeWords(uint8_t devAddr, uint8_t regAddr, uint8_t length, uint16_t *data);

                     ^

MPU6050.cpp:3301:68: error: invalid conversion from 'int16_t* {aka short int*}' to 'uint16_t* {aka short unsigned int*}' [-fpermissive]

    I2Cdev::writeWords(devAddr, SaveAddress + (i * shift), 1, &Data );

                                                                    ^

In file included from sketch\MPU6050.h:40:0,

                 from sketch\MPU6050.cpp:38:

I2Cdev.h:124:21: error: initializing argument 4 of 'static bool I2Cdev::writeWords(uint8_t, uint8_t, uint8_t, uint16_t*)' [-fpermissive]

         static bool writeWords(uint8_t devAddr, uint8_t regAddr, uint8_t length, uint16_t *data);

                     ^
[... 61 lines cut ...]
I2Cdev.h:115:23: error: initializing argument 4 of 'static int8_t I2Cdev::readWords(uint8_t, uint8_t, uint8_t, uint16_t*, uint16_t)' [-fpermissive]

         static int8_t readWords(uint8_t devAddr, uint8_t regAddr, uint8_t length, uint16_t *data, uint16_t timeout=I2Cdev::readTimeout);

                       ^

MPU6050.cpp:3325:42: error: invalid conversion from 'int16_t* {aka short int*}' to 'uint16_t* {aka short unsigned int*}' [-fpermissive]

  I2Cdev::readWords(devAddr, 0x13, 3, Data);

                                          ^

In file included from sketch\MPU6050.h:40:0,

                 from sketch\MPU6050.cpp:38:

I2Cdev.h:115:23: error: initializing argument 4 of 'static int8_t I2Cdev::readWords(uint8_t, uint8_t, uint8_t, uint16_t*, uint16_t)' [-fpermissive]

         static int8_t readWords(uint8_t devAddr, uint8_t regAddr, uint8_t length, uint16_t *data, uint16_t timeout=I2Cdev::readTimeout);

                       ^

exit status 1
invalid conversion from 'int16_t* {aka short int*}' to 'uint16_t* {aka short unsigned int*}' [-fpermissive]
```

i3130002, 2019-07-31:
I published a pull request that solves my errors in [here](https://github.com/jrowberg/i2cdevlib/pull/464). Not being sure that It was relevant to this issue in the first place.