jrowberg/i2cdevlib #185: Why sleep in getMotion9?

judepereira, 2015-08-16:
Instead of the current implementation, where the code first sets some bypass flag, then enable the mag, why not do this stuff during initialize() and read the mag data directly?

ist165351, 2015-12-16:
I would very much like to see this answered. Is there really a good reason for the sleep inside the read method? I intend to use the MPU9150 for brushless motor control, and I would like to run it at a few hundred Hz. The 10ms sleep inside the read method makes this impossible of course

ohsm88, 2015-12-16:
NAVER - http://www.naver.com/
--------------------------------------------

ohsm88@naver.com 님께 보내신 메일 <Re: [i2cdevlib] Why sleep in getMotion9? (#185)> 이 다음과 같은 이유로 전송 실패했습니다.

--------------------------------------------

받는 사람이 회원님의 메일을 수신차단 하였습니다. 

--------------------------------------------