jrowberg/i2cdevlib #193: I have a question about the MPU6050_DMP6.ino

a1090200, 2015-10-27:
HI everyone:
 I modified this sample code for some purpose and I found that my process crash usually.

In the [MPU6050_DMP6.ino Line:275](https://github.com/jrowberg/i2cdevlib/blob/master/Arduino/MPU6050/Examples/MPU6050_DMP6/MPU6050_DMP6.ino#L275) , I dicovered that the varible "mpuIntStatus"  should do bitwise operation with "0x02"  but  not "0x01". Am I right?

P.S. After I modified it ,my process ran stably.

Reference: MPU-6050 Register Map document section 4.16

shiehnpin, 2015-11-05:
After reading source code(MPU6050.h), I guess that normal data ready cause bit 0 == true and DMP data ready cause bit 1 and 0 both true (reverse-engineer). So you can just masking by 0x01 since either one will make bit 0 == true.

a1090200, 2015-11-05:
Hi shiehnpin:
    Thanks for you reply.
    This information is really helpful! I'll check the  source code and figure out what's happen to my original code. Thanks a lot.

shiehnpin, 2015-11-05:
In my experience, make sure FIFO is not overflow if you crash frequently.

a1090200, 2015-11-05:
I found a strange phenomenon that the  fifoCount can be greater than  1024 sometimes.In my case,the number is 27137. So the code in  [ MPU6050_DMP6.ino Line:269 ] (https://github.com/jrowberg/i2cdevlib/blob/master/Arduino/MPU6050/Examples/MPU6050_DMP6/MPU6050_DMP6.ino#L269)may cause my program crash.

shiehnpin, 2015-11-06:
27137 is a strange number. According to datasheet, FIFO counter is a 2-bytes register which means maximum can be 65536 or 32768. I suggest you print out Interrupt status every time, and pay attention to status while crash happened.

kobaj, 2016-01-11:
I noticed the same issue and checked the manual, a1090200 you appear to be correct.
https://store.invensense.com/Datasheets/invensense/RM-MPU-6000A.pdf

The bitwise operation should be with `0x01` and not `0x02`. I was experiencing multiple FIFO overflows until I switched to the former.

ohsm88, 2016-01-11:
NAVER - http://www.naver.com/
--------------------------------------------

ohsm88@naver.com 님께 보내신 메일 <Re: [i2cdevlib] I have a question about the MPU6050_DMP6.ino (#193)> 이 다음과 같은 이유로 전송 실패했습니다.

--------------------------------------------

받는 사람이 회원님의 메일을 수신차단 하였습니다. 

--------------------------------------------