jrowberg/i2cdevlib #196: MPU6050 - Wakeup Cycle Frequency

tony-gosu, 2015-11-03:
In the file MPU6050.h, line 417, the wakeup cycle frequency is defined as such:

#define MPU6050_WAKE_FREQ_1P25      0x0
#define MPU6050_WAKE_FREQ_2P5       0x1
#define MPU6050_WAKE_FREQ_5         0x2
#define MPU6050_WAKE_FREQ_10        0x3

But in the datasheet and the register sheet , the frequency is defined as

0 = 1.25Hz
1 = 5Hz
2 = 20Hz
3 = 40Hz

If I sit in a tight loop to see how often the accelerometer low power mode updates per this wake up frequency, the values match the datasheet and register sheet as described above, not what the library currently defines. Am I missing something or is this a bug that needs to be changed?

jrowberg, 2015-11-03:
The register map that I originally built this code from was v2.0 from May of 2011. The may have changed this in the official release, on non-engineering-sample parts. If your observations match the updated doc, then the code should probably be changed.

storyinvisible, 2019-07-31:
Hi, what should I do if I want to keep the frequency at 20 z?