jrowberg/i2cdevlib #197: MPU6050 library is not cmake-friendly

ibalashov, 2015-11-05:
@see https://github.com/queezythegreat/arduino-cmake/issues/105

Apparently in cmake the include file must be named after the directory.
In MPU6050 library one has to include "MPU6050_6Axis_MotionApps20.h" which is not compilable by cmake.

maartenverbandt, 2015-11-21:
As people on the arduino forum (http://forum.arduino.cc/index.php?topic=351281.15) pointed out, the header file should not include definitions of the functions. This results in serious problems when compiling to .o and linking afterwards. What you can do however is put the MPU6050_6Axis_MotionApps20.h in the MPU6050.cpp file, with an additional include guard, such that it only gets built when the motion apps are requested.
From a transparency point of view, it is probably better to make a separate dmp class, inheriting from the mpu6050 device class, in order to access its dmp functions.

jrowberg, 2021-09-28:
This should be fixed with the most recent commit.

closed 2021-09-28 by jrowberg