jrowberg/i2cdevlib #200: Do I need use a Kalman Filter or others using a DMP mode?

opvini, 2015-11-11:
Hi everyone,
when I use a DMP mode, I saw the values are filtered. So, I don't need to use others filters (Kalman or Complementary) after read data from DMP mode?

Thanks a lot

hgvk94, 2015-11-11:
Hi
You need not implement any fusion algorithms in your code.The DMP does this for you. the guys who manufacture these(http://www.invensense.com/products/motion-tracking/6-axis/mpu-6050/ ) havent revealed what algorithm they have used inside the DMP to accomplish this, but it does provide pretty stable results. 
See this link for a comparison in performance: http://www.geekmomprojects.com/mpu-6050-redux-dmp-data-fusion-vs-complementary-filter/

Cheers

opvini, 2015-11-11:
Very good!
Thanks a lot @hgvk94

closed 2021-09-26 by jrowberg