jrowberg/i2cdevlib #202: problem with dmpInitialize

Tchunk92, 2015-12-04:
Hi,
I'm stuck at the dmpInitialize function. I tried to see were the problem was. It seems that the function writeMemoryBlock can't do her job except with verify and useProgMem forced to true. What may be the cause of this?

Thanks,
Bruno

opvini, 2015-12-04:
Hi Bruno, do you tried to use the MPU6050_DMP6 Example?

Att,
Vinícius.

Tchunk92, 2015-12-08:
Hi,
Yes, with these modifications:
http://arduinoandmpu6050.blogspot.fr/2013/11/arduino-due-wiht-mpu6050.html

Thanks for answering,

Bruno

tplooker, 2016-03-19:
Anyone have an answer to whats causing this as I'm having the same problem?

opvini, 2016-03-20:
After a lot of testes, I gave up to use DMP mode and I sugest you to do the same.

tplooker, 2016-03-21:
Can anyone confirm they have got the MPU 9150 to work in dmp mode with this library?