jrowberg/i2cdevlib #206: Would you accept a PR for AK8963 support?

loopj, 2016-01-03:
Hey Jeff, I'm working on a project which includes the AK8963 magnetometer, and was wondering if you would accept a PR for support?

The AK8963 is substantially similar to the AK8975 which is already supported by i2cdevlib, but has a couple of extra registers, mainly because the resolution of the AK8963 is configurable.

Would you be interested if I submitted a PR for AK8963 support? Also, what would be your preferred approach - copying AK8975.cpp/h or editing those files directly and have AK8975.cpp/h support both chips?

Thanks!

ohsm88, 2016-01-03:
NAVER - http://www.naver.com/
--------------------------------------------

ohsm88@naver.com 님께 보내신 메일 <[i2cdevlib] Would you accept a PR for AK8963 support? (#206)> 이 다음과 같은 이유로 전송 실패했습니다.

--------------------------------------------

받는 사람이 회원님의 메일을 수신차단 하였습니다. 

--------------------------------------------

udawat, 2016-01-03:
I do not speak for Jeff but from past experience on i2cdevlib, it is suggested that you create a new folder for AK8963.

That way, a newcomer will be saved from confusion and previous projects created using AK8975 would not break.

From what i remember, MPU-6050 and MPU-9150 are also similar but now reside in their own folders.

ohsm88, 2016-01-03:
NAVER - http://www.naver.com/
--------------------------------------------

ohsm88@naver.com 님께 보내신 메일 <Re: [i2cdevlib] Would you accept a PR for AK8963 support? (#206)> 이 다음과 같은 이유로 전송 실패했습니다.

--------------------------------------------

받는 사람이 회원님의 메일을 수신차단 하였습니다. 

--------------------------------------------

jrowberg, 2016-01-03:
Thanks, both of you. @udawat is correct; the preferred approach is to have a new fileset for every distinct device. This will result in a lot of duplicated code, but the only way this would be inefficient in the actual project is if the developer is using both the AK8975 and AK8973, which would be very strange indeed.

I'd definitely accept a good PR for the AK8973.

closed 2016-01-03 by jrowberg