jrowberg/i2cdevlib #209: i2cdevlib license?

loopj, 2016-01-10:
Hey @jrowberg, thanks again for building such an awesome library. I was wondering if you could tell me what license the library is released under, or maybe even add a LICENSE.txt file to the repo?

ohsm88, 2016-01-10:
NAVER - http://www.naver.com/
--------------------------------------------

ohsm88@naver.com 님께 보내신 메일 <[i2cdevlib] i2cdevlib license? (#209)> 이 다음과 같은 이유로 전송 실패했습니다.

--------------------------------------------

받는 사람이 회원님의 메일을 수신차단 하였습니다. 

--------------------------------------------

jrowberg, 2016-01-11:
All of the code is under the MIT license; this should be present in comment blocks throughout all source files, but I will add a license file soon as well.

closed 2016-01-11 by jrowberg

loopj, 2016-01-11:
Thanks Jeff, not sure how I missed that!