jrowberg/i2cdevlib #216: MPU6050_DMP6:  OUTPUT_READABLE_YAWPITCHROLL: p and r limited to +-90 degrees, y working correctly

beng27, 2016-03-02:
Thanks for this great library.  I'm using an epalsite.com SY1547 board with the MPU6050,  and the stock example arduino sketches.

I have an issue that OUTPUT_READABLE_YAWPITCHROLL  gives ranges from -180 to 180 for yaw but for both pitch and roll, as you rotate the board it approaches 90 degrees then drecreases again.

E.g. if I rotate board around x axis, roll reading  rises from 0 to 90  (at 90degrees) then decreases 90 to 0 (at 180 degrees) then 0 to -90 (at 270 degrees) then -90 to 0 (at 360 degrees)

I've checked the web for this and seen other people mentioning exactly the same issue, but no replies or solutions. I have  calibrated the MPU but not solved it.

beng27, 2016-03-06:
I've been working on this and I offer a better coding of the  conversion function in MPU6050_6Axis_MotionApps20.h starting on line 645. 
 
This gives the correct readings of +- 180 degrees in all 3 axes in the stock demo code
 
This was figured out using info from http://www.euclidean...Euler/index.htm
and crucially the discussion at http://stackoverflow...the-orientation
 
Please note: 1) Normally I develop Filemaker Databases and I  really don't understand quaternions either!  2) I've only been using the Ardiono for a week or so and never seen C++ before, it may be that my code is not efficient. However I think the logic is sound.
 
Ben
 
_______
int8_t    sign;

uint8_t MPU6050::dmpGetYawPitchRoll(float *data, Quaternion *q, VectorFloat *gravity) {
    // p & r need to check for >90 rotation and correct for gravity polarity reversal
    if (gravity -> z > 0) {sign = 1;} else {sign = -1;}
    // yaw: (about Z axis)
    data[0] = atan2(2*q -> x*q -> y - 2*q -> w*q -> z, 2*q -> w*q -> w + 2*q -> x*q -> x - 1);
    // pitch: (nose up/down, about Y axis)
    // adding tiny fraction of x2 or y2  avoids gimbal lock? at cost of slight accuracy
    data[1] = atan2(gravity -> x , sign * sqrt(0.001*gravity -> y*gravity -> y + gravity -> z*gravity -> z));
    // roll: (tilt left/right, about X axis)   
    data[2] = atan2(gravity -> y , sign * sqrt(0.001*gravity -> x*gravity -> x + gravity -> z*gravity -> z));
  return 0;
}

eadf, 2016-04-17:
I second that, I don't understand quaternions either!

#222 had another fix for the roll & pitch problem and I checked in those to the 'develop' branch for easy testing.

This solution is a bit different from yours, so it would be nice if we could find consensus on what solution to pick. Maybe your solution is better with regards to gimbal lock?

P.S. your links seems to be corrupted.

maar111, 2017-02-06:
hello, any news about this problem?

zouden, 2018-06-26:
Hi guys,
I had the same problem and found this solution from @beng27 on the i2cdevlib forum and then again here. I saw that a different fix (#222) has been added to the develop branch. I have now tested both and it's clear that #222 only solves the problem when the roll axis is basically horizontal. The code here from @beng27 works at any angle, and should be adopted in favour of #222. 

The code can be found without formatting errors here: https://www.i2cdevlib.com/forums/topic/24-roll-and-pitch-angles-ranges/

beng27, 2018-06-27:
Hi!  please beware that while I do still think my solution is correct,  I am definitely not an expert and arrived at it by educated guessing rather than full mathematical rigour.

Not sure of the right way to feed this info back as I am not a github person; taught myself a bit of arduino programming to do a project and that’s all.

best wishes
Ben