jrowberg/i2cdevlib #217: interrupt dmp

marcodatola, 2016-03-22:
Hi am looking for a way to completely stop the dmp or a way to stop it firing interrupts, if a certain event occurs. How can I achieve such thing? Thanks in advance

eadf, 2016-03-25:
Assuming you are talking about the MPU6050 driver...

If you call`void setIntEnabled(uint8_t enabled);` you can control the interrupts it fires.

`mpu.setIntEnabled(0)` will turn them all off.

To re-enable them you can call:

`mpu.setIntEnabled(1<<MPU6050_INTERRUPT_FIFO_OFLOW_BIT|1<<MPU6050_INTERRUPT_DMP_INT_BIT);`

`MPU6050_9Axis_MotionApps41.h` and `MPU6050_6Axis_MotionApps20.h` uses `mpu.setIntEnabled(0x12);` (same result)

When you re-enable interrupts you will get a fifo overflow (naturally)

marcodatola, 2016-03-30:
hi, thanks for the reply. 

not sure what you mean by driver? I'm interfacing mpu6050 IMU with particle photon MCU.

could you advise how to deal with the FIFO overflow please?

eadf, 2016-03-30:
I'm sorry if I was unclear. By driver I mean the piece of 'glue' software that is specific to the device, the class that calls the generic methods of i2cdevlib.`MPU6050.cpp`and`MPU6050.h`in your case.

I don't have much experience with the mpu6050 device, but my instinctual reaction to your problem with the fifo queue would be to read (and throw away) enough samples so that you know that you are getting fresh data.