jrowberg/i2cdevlib #222: YawPitchRaw algorithm mistaken

yonoodle, 2016-03-26:
Thank you for this WONDERFUL lib.

buy the "roll" algorithm in the dmpGetYawPitchRoll should be 
data[2] = atan2(gravity -> y , gravity -> z);

(at least one of the pitch or roll should be as simple as that, and only one uses sqrt)
see complete derivation 
https://cache.freescale.com/files/sensors/doc/app_note/AN3461.pdf

about extending pitch and roll from degenerate -90~90  to non degenerate -180~180
i simply used the code below.

uint8_t MPU6050::dmpGetYawPitchRoll(float *data, Quaternion *q, VectorFloat *gravity) {

    
   
    // yaw: (about Z axis)
    data[0] = atan2(2*q -> x*q -> y - 2*q -> w*q -> z, 2*q -> w*q -> w + 2*q -> x*q -> x - 1);

    //pitch
    
    data[1] = atan2(gravity -> x , sqrt(gravity -> y*gravity -> y + gravity -> z*gravity -> z));
    //roll
    data[2] = atan2(gravity -> y , gravity -> z);

    if(gravity->z<0)
    {
        if(data[1]>0)
        {  data[1] = 3.1415926 - data[1] ; }
        else { data[1] = -3.1415926 - data[1] ; }
    }

    
   

    
  return 0;
}

eadf, 2016-04-01:
If we had a development branch I could check-in quick fixes like this. The community could then take part in the testing. What do you think @jrowberg? You will still have the final say on what gets merged into`master`.

jrowberg, 2016-04-01:
@eadf: I'd be perfectly happy with that approach. Back when I first created the repository, I knew very little about Git and best practices, so doing that at the time didn't even occur to me.

eadf, 2016-04-16:
I've just pushed a updated `MPU6050::dmpGetYawPitchRoll()` implementation to the 'develop' branch.
The new code (thanks @yonoodle) can detect if the device is upside down. The old code could not.

You can compare the old `dmpGetYawPitchRoll()` with this (temporary) #define:

```
#define USE_OLD_DMPGETYAWPITCHROLL
#include "MPU6050_6Axis_MotionApps20.h"
```

eadf, 2021-09-23:
#216 seems to be the better choice.
I think the develop branch can be archived now, as it is badly out of date.

jrowberg, 2021-09-23:
> I think the develop branch can be archived now, as it is badly out of date.

I don't see a way to truly archive a branch, only delete. Is there anything necessary to keep? Should I tag the develop HEAD and then just delete the branch? (Or would that delete the tag and everything with it?)

eadf, 2021-09-23:
Hm, I don't know what the best solution is. Maybe just leave it for a few days so we can evaluate the content

eadf, 2021-09-26:
Well, this solution is in master MPU6050_6Axis_MotionApps_V6_12.h now. (Can't find #216)
Close?