jrowberg/i2cdevlib #227: fifoCount is not set to zero automatically after mpu.resetFIFO() , is this designed intentionally?

yonoodle, 2016-04-14:
Dear jrowberg et all ,

because i have to design the code to access the mpu6050 data sometimes but not all the time ,when I use resetFIFO() method , i have to manually set fifoCount to zero for the code to prevent overflow , just like code below.

--------------------
.
.
.
 OBJ_1.resetFIFO();
  fifoCount_OBJ_1 = 0;
.
.
.
----------------------

is this designed intentionally for some use? or the fifoCount zeroing can be merge to the resetFIFO method automatically for more intuitive use?

eadf, 2016-04-17:
I think you are correct when you say that fifoCount should be updated after each resetFIFO().

About `fifoCount` zeroing could be merged with the`resetFIFO()`method: 
Are you suggesting that fifoCount should be moved to `MPU6050::fifoCountCached` (or something similar)?  `MPU6050::getFIFOBytes()` and `MPU6050::resetFIFO()` should alter the cached value.

yonoodle, 2016-04-18:
if fifoCount is not meant to be manipulated for some special usage , setting it private might be a neat way, but it seems we need to use it for some custom condition deciding to execute some loops or not  , maybe it can be sent as a parameter to some of the corresponding methods?