jrowberg/i2cdevlib #231: Helpme in getting I2Cdevlib for Cortex M3 to getting data from MPU6050

473srikanth, 2016-05-03:
I want to interface MPU6050(GY521) to cortex M3 (1343 board). Can any one Helpme , where can i find the I2Cdev library for Cortex M3??

eadf, 2016-05-03:
Excuse my ignorance, but what platform does the cortex M3 on a 1343 use?
_Arduino Due_ uses a Cortex M3 (obviously Arduino)
_Leaflabs Maple_  uses [libmaple](https://github.com/leaflabs/libmaple) or a variant of [Arduino](https://github.com/rogerclarkmelbourne/Arduino_STM32)

If the underlying API is Arduino the i2cdevlib/Arduino code should work afaik

eadf, 2016-05-04:
Excuse my ignorance, but what software platform does the _Cortex M3 on a 1343_ use?
_Leaflabs Maple_  runs on an ARM Cortex-M3 and it uses [libmaple](https://github.com/leaflabs
/libmaple) or a variant of [Arduino](https://github.com/rogerclarkmelbourne/Arduino_STM32)
_Arduino Due_ runs on a Cortex M3 (obviously Arduino)

If the underlying API is Arduino the i2cdevlib/Arduino code should work afaik

473srikanth, 2016-06-10:
Hello eadf, thanks  for reply. I'm using NXP LPC1343 microcontroller based  Board. It's not Arduino or Leaflabs Maple board. Its a local made board.    where can i find the I2Cdev library for this controller??