jrowberg/i2cdevlib #232: I2CXL-MaxSonar-EZ Support

NCharabaruk, 2016-05-04:
Hello,

I am attempting to use an I2CXL-MaxSonar-EZ2 (MB1222) ultrasonic sensor and an MBU6050 IMU on a single Arduino. The MB1222 code that is provided on the datasheet (http://www.maxbotix.com/documents/I2CXL-MaxSonar-EZ_Datasheet.pdf) uses softi2cmaster and I would like to convert it to i2cdevlib with fastwire. I am having no luck getting it to work as I have little experience with this kind of programming. Would it be possible to get support for this device?

Thanks

eadf, 2016-05-04:
Hm, that device is a little bit odd as it does not use any register addresses when calling `start_sensor()` and `read_sensor()`. 

I don't have access to a MaxSonar-EZ so I can't do the usual trial and error tests, but maybe this simple code works.

untested, uncompiled, error-prone-meta-code:
```
bool start_sensor(uint8_t devAddr) {
   //bool writeBytes(uint8_t devAddr, uint8_t regAddr, uint8_t length, uint8_t *data);
  return I2Cdev::writeBytes(devAddr, 81, 0 /* yes, zero */, NULL);
}

uint16_t read_sensor(uint8_t devAddr) {
   uint16_t rv;
   //static int8_t readWord(uint8_t devAddr, uint8_t regAddr, uint16_t *data); 
   // this will send an superfluous byte (0), but maybe the device will accept it any ways.
   I2Cdev::readWord(devAddr, 0, &rv);
   return rv;
}
```

NCharabaruk, 2016-05-06:
Hello again,

Sorry to bother you so much. I tried using this code with your PR of I2Cdevlib. I first tried:

`void loop() 
{
  Serial.print("count\t");
  Serial.print(count);
  Serial.print("\t");
  Serial.print("range\t");
  delay(100);
  Serial.println(read_sensor(224));
  count++;
}`
 
which resulted in 

> count	0	error		range	count	0	error		range	29440
count	1	error		range	

before locking up. I then added error checking:

`void loop() 
{
  int error = 0;
  Serial.print("count\t");
  Serial.print(count);
  Serial.print("\t");
  Serial.print("error\t");
  Serial.print("\t");
  
  error = start_sensor(224);
  Serial.print(error);
  Serial.print("\t");

  if(error == 0)
  {
    Serial.print("range\t");
    delay(100);
    Serial.println(read_sensor(224));
  }
  else
  {
    Serial.println("There was an error");
  }
  count++;
}`

which resulted in

> count	1	error		1	There was an error
count	2	error		1	There was an error
count	3	error		1	There was an error
count	4	error		1	There was an error
count	5	error		1	There was an error
count	6	error		1	There was an error
count	7	error		1	There was an error
count	8	error		1	There was an error
count	9	error		1	There was an error

and kept going until I closed the serial monitor.

eadf, 2016-05-06:
Ok, here are a few things you can test/check:

It is possible to run the MPU6050 on one set of wires/pins (hardware i2c) and the MaxSonar on another (software i2c). Did you consider this?

The code i suggested above is, by second thought, unlikely to work. The functions start_sensor() and read_sensor() are just one single 'read register operation' in point of view of the device. 

I suggest that you modify the MaxSonar softi2cmaster code that you know works in such a way that you can use it together with i2cdevlib running in softi2cmaster mode. (You will need i2cdevlib for the MPU6050 later)
SoftI2CMaster.h can only be #included once in a program, so I made the class SoftI2CMasterWire that exports the functions of SoftI2CMaster.h so other parts of the program can use them.
Your modified code needs to call the SoftI2CMasterWire methods instead of directly calling the methods of  SoftI2CMaster.h. 
E.g. Here is the converted read_sensor() method:

```
// remove #include <SoftI2CMaster.h>  
int read_sensor(byte bit8address) {
  boolean errorlevel = 0;
  int range = 0;
  byte range_highbyte = 0;
  byte range_lowbyte = 0;
  bit8address = bit8address | B00000001;

  errorlevel = !SoftI2CMasterWire::i2c_start(bit8address) | errorlevel;
  range_highbyte = SoftI2CMasterWire::i2c_read(0);
  //Read a byte and send an ACK (acknowledge)
  range_lowbyte = SoftI2CMasterWire::i2c_read(1);
  //Read a byte and send a NACK to terminate the transmission
  SoftI2CMasterWire::i2c_stop();
  range = (range_highbyte * 256) + range_lowbyte;
  //compile the range integer from the two bytes received.
  if (errorlevel) {
    return 0;
  }
  else {
    return range;
  }
}
```

Note that the device address 224 in the example is bit-shifted. i2cdevlib assumes un-shifted addresses. However, this has probably no implications on your current problem.

NCharabaruk, 2016-05-06:
Ok, I will give this a shot and let you know how it goes. 

> It is possible to run the MPU6050 on one set of wires/pins (hardware i2c) and the MaxSonar on another (software i2c). Did you consider this?

I was unaware this was a possibility so I have not looked into it.

NCharabaruk, 2016-05-06:
SUCCESS! 

For anyone else that might look at this later, the 'i2c_' need to be removed from the sample code above as they are not in the class (ie. SoftI2CMasterWire::start(bit8address) rather than SoftI2CMasterWire::i2c_start(bit8address))

closed 2016-05-06 by NCharabaruk

eadf, 2016-05-06:
Nice.

I could use another set of eyes on the #220 PR, so don't hesitate to give some feedback if you spot anything that needs fixing.