jrowberg/i2cdevlib #233: Simplify MPU6050_DMP6 code to use with others methods - Arduino

jesuspk157, 2016-05-12:
Hi.

I think the MPU6050_DMP6 code it's the best code for that sensor because it give x, y and z angles with precision.

But if I want to use the MPU sensor with others methods (to move servos, calculate values, etc.) the code don't give data or stoped; I don't understand how work all your code but it looks that the send and received data have a determine time to be sended else you don't have values.

There are any way to simplify the work to obtain the x, y and z angles with out too much problems?

I see you have code for raw values, the code it's simple but I don't know which operations to do for obtain the angles.

I hope you can help me, sorry for my english. Thank you.

eadf, 2016-05-13:
> But if I want to use the MPU sensor with others methods (to move servos, calculate values, etc.) the code don't give data or stoped;

There is a MPU6050_DMP6 bug in the master branch that can cause such behaviour. It is fixed in the 'develop' branch. So the easiest thing would be if you tested your code against the develop branch and see it the problem persist.

> I don't understand how work all your code but it looks that the send and received data have a determine time to be sended else you don't have values.

Yes, at default speed (100Hz) the loop is quite time sensitive, but you can change the FIFO rate of the DMP in MPU6050_6Axis_MotionApps20.h.

```
const unsigned char dmpConfig[MPU6050_DMP_CONFIG_SIZE] PROGMEM = {
.......
    0x02,   0x16,   0x02,   0x00, 0x01                // D_0_22 inv_set_fifo_rate

    // This very last 0x01 WAS a 0x09, which drops the FIFO rate down to 20 Hz. 0x07 is 25 Hz,
    // 0x01 is 100Hz. Going faster than 100Hz (0x00=200Hz) tends to result in very noisy data.
    // DMP output frequency is calculated easily using this equation: (200Hz / (1 + value))

    // It is important to make sure the host processor can keep up with reading and processing
    // the FIFO output at the desired rate. Handling FIFO overflow cleanly is also a good idea.
};
```

> There are any way to simplify the work to obtain the x, y and z angles with out too much problems?

I fail to see how that could be done, since the DMP code needs to react to a external interrupt. And it's not advisable (is it even possible?) to call i2c methods in the interrupt function. So the algorithm must be similar to what we're already doing: 
```
interruptFunction(){
  when MPU6050 says data is ready we set 'data ready' bit
}

loop() {
  as fast as possible, check 'data ready' bit
    is set: acquire sensor values via i2c
    else: do something else that isn't too time consuming
}
```
I'll have to get back to you regarding how to get x, y and z angles out of the non-DMP example code.

udawat, 2016-05-13:
These are all excellent recommendations. It may be possible that the problem is from the hardware side. @jesuspk157 Can you confirm how you are powering your servo motors? These require high currents and should be supplied power with an external power supply (usually 4V to 6V) and not the dev board itself. Grounds from the servo motors, sensor and dev board should all be connected.

Also, are you using the processor interrupt for connecting to something else other than to MPU-6050?
Anyways, if you do not want to use DMP, you can get useful data in roll, pitch, yaw form by using Mahony/Madgwick algorithms - [Madgwick link](http://www.x-io.co.uk/open-source-imu-and-ahrs-algorithms/). But, it will take up more code memory and is processor intensive.