jrowberg/i2cdevlib #236: I2Cdev & ADXL345 compilation error

dblanm, 2016-05-20:
I am trying to compile a code using I2Cdev.h and ADXL345.h and it gives me the following error:

I2Cdev.h:46:1: error: unknown type name ‘class’
 class I2Cdev {
 ^
I2Cdev.h:46:14: error: expected ‘=’, ‘,’, ‘;’, ‘asm’ or ‘__attribute__’ before ‘{’ token
 class I2Cdev {
              ^
In file included from IMU.c:12:0:
ADXL345.h:163:1: error: unknown type name ‘class’
 class ADXL345 {
 ^
ADXL345.h:163:15: error: expected ‘=’, ‘,’, ‘;’, ‘asm’ or ‘__attribute__’ before ‘{’ token
 class ADXL345 {

I haven't modified the code. Has anyone had this problem before?

Thank you for your time!

eadf, 2016-05-20:
First some mandatory questions..
* What is your target platform?
* What happens when you try to compile the ADXL345_raw example?
* Are you using the current i2cdev master branch?

Judging from the error text I would suggest you to look at what happens just before `#include <I2Cdev.h>` in your sketch. Try to temporarily move the failing #include to the top or the bottom of the #include block. If compiler complains about something else the problem is outside of I2Cdev.h.

dblanm, 2016-05-21:
My target platform is Raspberry Pi 2B, actually I compiled ADXL345 example without problems. Thank you for the help!

dblanm, 2016-05-21:
The code I am using is in [https://github.com/dblanm/IMU_gy85](url)

eadf, 2016-05-21:
I think the main problem is the macro:
`#define ADXL345 0x53` in sensors.h

eadf, 2016-05-21:
Are you using g++ when compiling your code? ADXL345.h and I2Cdev.h are C++ files.

dblanm, 2016-05-22:
In fact I was compiling using gcc. I just used the i2c-dev.h and adapted your libraries to mine. Thank you for your help!

closed 2016-05-22 by eadf