jrowberg/i2cdevlib #238: gravity integer factor error?

yonoodle, 2016-05-23:
in MPU6050_6axis_motionApps20.h , around line 660 , in the dmpGetLinearAccel() ,
it multiplys 8192 for 2g sensitivity,

but for 2 byte raw data of acceleration ,  -32767~32767 for +- 2g , 1g is for 16384 ,
i tried the correction , and works. which match the raw data when MPU6050 is still and lay horizontal.

shouldn't this be checked? it seems a big bug?

eadf, 2016-05-24:
I'm not entirely up to speed with the subject, but is this related to this [commit+comment](https://github.com/jrowberg/i2cdevlib/commit/30f6289bebacbf3634910277850b2b4452cd4dc0)?

yonoodle, 2016-05-24:
yes, but i think the correct value is not 8192 but 16384 , isn't it?  the raw data shows 1g = 16384 for a still MPU6050.  and the gravity vector norm calculated by quaternion is 1 , which should be multiply 16384 to match the 1g raw data.

yonoodle, 2016-05-24:
well , i found i was wrong,  the dmpGetLinearAccel() uses dmpAccel() output value as input , which is indeed 8192 for 1g other than 16384 for 1g by getMotion6() method.

what is the pro and con between dmpAccel() and getMotion6()? the later one seems have higher resolution?