jrowberg/i2cdevlib #240: Connecting 2 MPU6050s with Arduino using only 1 MPU6050 SW Object

sidg-vr-geek, 2016-06-02:
Greetings,

My project requires connecting multiple MPU sensors and sending all rotation data to comp using USB.

I have the single sensor code working absolutely fine. Now i am hooking up 2 sensors. Idea is :

- Tie together the SDA, SCL, Vcc, GND and INT lines of both MPUs together.
- Connect the AD0 pins of MPUs to different Arduino pins.
- **Declare only 1 MPU6050 object** (as given by jeff rowberg) with devAddr = 0x69 (I dont want to maintain X no. of object for X no. of sensors. My project going ahead may need 8+ sensors and i can't have 8+ MPU objects in my Arduino)
- **Whenever any sensor needs to be Init / Read / Written to, enable that sensor by writing HIGH to AD0 pin, and writing LOW on all other AD0 pins of other sensors.**
- Once the particular sensor has been enabled, i use the normal mpu functions from my MPU6050 object which have been provided by Jeff.

Now while initializing, i enable sensor 1, call mpu.initialize, testconn, dmpInit.... etc... , then i enable sensor 2, and do the same. Same method is followed to collect data from all sensors.

Problem is, only Sensor 2 gives valid output. Sensor 1 gives corrupt values.

I debugged the problem a little bit, it seems only the sensor which i initalize the last (sensor 2 in above case) gives valid values.

I believe most probably when i init the second sensor using the single mpu object, it overwrites some SW vars in this object that were previously set by first sensor. Then when i try to access the Sensor 1 with this object, these overwritten values cause problems and snesor data comes out corrupt.

However i went through the whole MPU library to see if there are any private / public vars that are MPU specific and that Jeff's code might be saving. but i found nothing. It seems almost all the seful function straightaway call I2CDev Read/WriteBit. This implies as long as the correct device's AD0 is HIGH, and MPU's SW object's devAddr = 0x69, everything should work fine.

And yet it does not. Can someone help ?

Thanks

sidg-vr-geek, 2016-06-02:
Well my problem got solved. Turns out there was a  problem with MPU FIFO.

it seems that the sensor i init first had an extra 16 bytes in its FIFO (I have no idea why). My code logic was reading 42 bytes (dmp packet size) chunks and was leaving this 16 bytes as is. This was causing corruption.

Now i keep reading the data till fifo is empty. and everything is working fine.

closed 2016-06-02 by sidg-vr-geek

eadf, 2016-06-02:
I'm currently testing an [extended DMP6 example](https://github.com/eadf/MPU6050_DMP6_Multiple) that allows us to connect to several MPU6050. It uses the same AD0 technique as you are describing. I will upload this to the i2cdevlib develop branch as soon as I've done some more tests.

My findings so far is that the DMP interrupt will sooner or later (can be minutes or days) make Wire.h to freeze up. After removing the interrupt code the tests with two MPU6050 (FIFO rate 40Hz) with a 8Mhz pro mini will run for at least 48hours (that's when i shut down the 'puter).

Feedback is welcome

eadf, 2016-06-02:
I replaced the interrupts with timestamp based polling. The example will start polling the device a few ms before a new sample is expected to be generated.