jrowberg/i2cdevlib #249: How to connect a Mpu 6050 and a HMC5883L together ?

Zephyrixroxx, 2016-06-16:
Hey I am new to arduino ,so i dont really know how to connect two different sensors.I have seprate codes written for both ,but i dont know how to physically connect them ,i am thinking of maybe getting this to work and then getting a mpu9150 ,but i wanted to that as the last thing.Please help me .Thank you in advance

icefreesky, 2023-07-23:
use module gy-87  Mpu 6050 and a HMC5883L one board

ZHomeSlice, 2023-07-23:
MPU9150  has an MPU6050 and a AK8963 magnetometer internal and I am not impressed with the AK8963.
[comparison between AK8963 vs HMC5883L](http://ejtech.blogspot.com/2014/03/mpu9150-vs-hmc5883l-trade-off-analysis.html) 
So the 9150 is just a fancy MPU 6050 with an i2c link internally to the AK8963. this also requires an external library to integrate the AK8963 which is not available with Jeff's MPU6050 library. Your choice to use the HMC5883L  magnetometer is a good one. you can still use the MPU9150 as an MPU6050. they will program exactly the same.

Z

icefreesky, 2023-07-24:
> MPU9150  has an MPU6050 and a AK8963 magnetometer internal and I am not impressed with the AK8963.
> [comparison between AK8963 vs [...]

Have you come across a ready-made library or a solution to connect mpu6050 + hmc5883l and get 9 axes using dmp like mpu9250?

ZHomeSlice, 2023-07-25:
Just like the MPU9150 The MPU9250 and 9255 have a magnetometer attached to the secondary I2C bus inside the chip. The 9250 has an MPU6500 which is a improved version of the MPU6050 The magnetometer has a better resolution also.  The eDMP firmware is identical for all four chips. The only major difference is where the accelerometer offsets are stored and Jeff's library handles that automatically. InvenSense has provided within their development kit that can be found on their website a pre-compiled library that could possibly run within the Arduino architecture. Adding the library for the Uno is not possible as there's no memory left to include that. I have tried experiments with including that library but I could not get it to compile using the Arduino IDE. If you'd like to play with this library I suggestion is to go to the InvenSense website and register as a developer and then you can download their complete developer kit that would allow you to enable the 9° of freedom feature and try it out. If you do find a way to get the 9* of freedom feature to work with the Arduino IDE I would be grateful for some pointers.

Z