jrowberg/i2cdevlib #251: How to stop DMP from MPU6050 for some seconds

batata004, 2016-06-23:
Hi,

I am pretty sure Jeff's MPU6050 library is the best in the internet to retrieve ypr angles from MPU6050 using the DMP. It's a really good library but it's not easy to beginners cause the way the basic Arduino code to retrieve ypr from the DMP is a little complicated.

Anyway, after spending almost 3 months I am confident I understand very well the code and it works really fine. But I found a problem: sometimes I really need to execute some routine that can take up to 3 to 10 seconds and Jeff's code simply crashes. Sometimes Arduino freezes and no FIFO OVERFLOW error message is displayed, other times the error message is displayed. I know that using DMP relies on interrupt and my code must be really really really optimize in order to not hang arduino for too long.

So I decided that before doing my routine I should do mpu.setDMPEnabled(false) and after I finish my routine I should do mpu.setDMPEnabled(true);. But there is a problem: after I execute mpu.setDMPEnabled(true); the MPU6050 restarts its internal "calibration" algorithm and takes at least 5 to 8 seconds to get solid readings to ypr.

Doing the method above I can at least prevent fifo overflow or my arduino to freeze, but it creates another problem cause now my new readings (after mpu.setDMPEnabled(true);) are not precise anymore for at least 5-8 seconds.

Is there someway to ignore DMP readings without overflowing the buffer and at the same time doing something else so in the future I can start to pay "attention" again to DMP and keep reading its angles without any recalibration?

SirBeni, 2018-06-06:
Hey there,
really sorry for you that nobody could help you. I got a similar problem and would be interested in how you solved it - did you make it? And another short question: is it after every mpu.setDMPEnabled(true); that it takes 5-8 seconds for the readings to becom eprecise? Is this normal?
Best regards!

batata004, 2018-06-06:
About the second question yeap, everytime you enable DMP it has a sel calibration algorithm so it will always take around 6 seconds to stabilize. If you set correctly the offsets to the gyro/accel it may take even less, close to 2 seconds.

The only way I found to stop DMP was disabling it BUT when doing so your next readings for a few seconds will be completely noisy/random. I didnt find any other alternative so I have to live with that.

SirBeni, 2018-06-06:
Ok, thank you for your fast response!

paynterf, 2019-09-30:
Can you cut your lengthy code section into chunks and do a mpu.ResetFIFO() call in between chunks?