jrowberg/i2cdevlib #252: Serious bug at Jeff Rowberg library with MPU6050 at Arduino

batata004, 2016-06-24:
Sometimes my Arduino freezes or crashes when using JEFF ROWBERG library to read DMP data from MPU6050. It took me almost a full year to discover the problem and today I had some spare time and I tried to find where exactly arduino hangs/freezes.

I discovered it is usually when executin getFIFOCount() or getIntStatus(). I discovered that inserting many Serial.print with tags along my entire code so I could know exactly which line hanged. So I discovered everytime my arduino hanged/freezes it was at a line with getFIFOCount() or getIntStatus(). So I decided to take a close look at it.

Inside the file MPU6050.cpp there is a code like this:

uint16_t MPU6050::getFIFOCount() {
    I2Cdev::readBytes(devAddr, MPU6050_RA_FIFO_COUNTH, 2, buffer);
    return (((uint16_t)buffer[0]) << 8) | buffer[1];
}

I checked the function "readBytes()" inside I2Cdev.cpp and I found this:

I2Cdev::readBytes(uint8_t devAddr, uint8_t regAddr, uint8_t length, uint8_t *data, uint16_t timeout)

You see the problem? The function readBytes accepts one last parameter which is "timeout" but the function getFIFOCount uses readBytes without the timeout parameter! That's probably a bug cause sometimes I think the readBytes takes too long to execute and a TIMEOUT SHOULD BE set to avoid the function hanging out forever.

I am pretty sure maaaany MAAAANY other people alterady experienced this problem and spent a lot of time trying to understand what went wrong. Some people mistakenly believe it's related to bufffer overflow in the FIFO or because of some non optimized code. It could be the problem but MY PROBLEM FOR ALMOST A FULL YEAR has been this bug reported here.

I really hope someone fix this and add a timeout so nobody would spent entire months anymore seeking a problem that's is really hard to track an intermitent.

bishwa9, 2016-06-27:
This is my first time using the library and I think I am running into this issue as well. Did you figure out what timeout to give?

Thank You.

batata004, 2016-06-28:
No, it looks like this library support is dead. I think we should move to other alternatives cause this library has already many bugs documented and unresolved so it's really unreliable if you are gonna use this library to anything serious. Jeff Rowberg is not providing the good support he gave to this library in the old days so I think you would be better off this library

jrowberg, 2016-06-28:
Apologies for the lack of support. I've tried to migrate most of the effort to online forums and other willing (and often more capable) contributors as much as possible, since I can't devote the time to this that I used to. The best laid plans of mice and men, as it were...

However, in this case, I strongly doubt that it is the lack of a timeout argument that is causing the underlying issue. Arduino's Wire library **does not actually provide timeout functionality**. If you dig into the code, you'll see a blocking loop in the TWI calls. See this for example:

https://github.com/arduino/Arduino/issues/1476

Supplying a timeout in I2Cdevlib code would make zero difference, because there's no way to pass it to the underlying Wire/TWI implementation. The only way around this is to use a different timeout-capable I2C implementation. This is possible, but not simple.

I would recommend checking the SDA/SCL lines with a logic analyzer to see when/why the sensor stops responding. The blocking loop inside Arduino's Wire/TWI code should not hang as long as the sensor responds according to the I2C spec.

batata004, 2016-06-28:
@jrowberg I never wanted to offend you. I know you did this library for free and you are contributing a lot with the community providing it to general use. This library is really great but there are some bugs with it. When you have sometime I would like you to take a look at this:

http://forum.arduino.cc/index.php?topic=408851.0

For some reason your library is a little "unstable", when you change some lines of code it simply stops working. I also provide you below with a really simple code to use your great library to retrive DMP data from MPU6050 without using interrupt pin. It simply freezes arduino after some time. I already checked with an osciloscope and everything is working fine there is not reason (at the hardware) that could make this freeze happens. Could you please run the code below in your arduino and wait a few seconds? After around 10-30 seconds your arduino will freeze.

```
#include "I2Cdev.h"
#include "MPU6050_6Axis_MotionApps20.h"
#include "Wire.h"
MPU6050 mpu;
uint16_t packetSize;
uint16_t fifoCount;
uint8_t fifoBuffer[64];
Quaternion q;
VectorFloat gravity;
float ypr[3];

void setup() {

    Wire.begin();
    TWBR = 24;
    mpu.initialize();
    mpu.dmpInitialize();
    mpu.setXAccelOffset(-1343);
    mpu.setYAccelOffset(-1155);
    mpu.setZAccelOffset(1033);
    mpu.setXGyroOffset(19);
    mpu.setYGyroOffset(-27);
    mpu.setZGyroOffset(16);
    mpu.setDMPEnabled(true);
    packetSize = mpu.dmpGetFIFOPacketSize();
    fifoCount = mpu.getFIFOCount();

    Serial.begin(115200);

}

void loop() {

    while (fifoCount < packetSize) {

		//here comes my other code
        
        
        
      
        fifoCount = mpu.getFIFOCount();

    }

    if (fifoCount == 1024) {
    
        mpu.resetFIFO();
        Serial.println(F("FIFO overflow!"));
        
    }
    else{
    
    	if (fifoCount % packetSize != 0) {
        
        	mpu.resetFIFO();
            
		}
        else{
    
            while (fifoCount >= packetSize) {
            
                mpu.getFIFOBytes(fifoBuffer,packetSize);
                fifoCount -= packetSize;
                
            }    
        
            mpu.dmpGetQuaternion(&q,fifoBuffer);
            mpu.dmpGetGravity(&gravity,&q);
            mpu.dmpGetYawPitchRoll(ypr,&q,&gravity);        	
            
            Serial.print("ypr\t");
            Serial.print(ypr[0]*180/PI);
            Serial.print("\t");
            Serial.print(ypr[1]*180/PI);
            Serial.print("\t");
            Serial.print(ypr[2]*180/PI);
            Serial.println();
            
		}
   
    }

}
```

hgvk94, 2016-06-28:
Hello,

I used to have the same problems. Try running it without the serial
prints.It had worked for me.

cheers and regards

batata004, 2016-06-28:
I tried it right now and didnt work :( I even tried changing the baud rate and it also didnt work. I think it's not related to Serial but to something in the library cause there is never a buffer overflow when I run the above code in my arduino so the Serial is not causing a significant delay that could cause a problem/overflow. I am pretty sure is something related to MPU6050 DMP library.

batata004, 2016-06-29:
@jrowberg please, when you have a time try the code I gave you in your arduino with MPU6050. I am sure it's gonna freeze and I would like to know from you where is the bug of the library so I can fix it on my end.

batata004, 2016-07-05:
:( I wish jeff could find the bug in this code

yang614, 2016-07-07:
So the line `for (; Wire.available() && (timeout == 0 || millis() - t1 < timeout); count++) ` doesn't work even you specify a value to timeout?  Then why write (timeout == 0 || millis() - t1 < timeout) rather than removing it?

batata004, 2016-07-07:
@yang614  I changed the line as you said and the bug still happens, arduino freezes. I dont know for sure if it's a timeout problem. Could you please test the code I provide in your arduino amd mpu6050? Other people on forum.arduino.cc could also replicate this bug, that's why I came here to github but unfortunatelly jeff does not look available to help with this

yang614, 2016-07-09:
I changed, too. Yes ,it still happens.I am waiting Rowberg to explain it.

batata004, 2016-07-11:
yeap, same here. I really hope Rowberg replies us here.

yang614, 2016-07-11:
Do you use mag?　I use the HMC5883L with mpu6050 to read the mag raw data and read the pitch yaw roll by DMP is fine. But when I use the mpu9150 to read the the mag data by DMP only , it keeps overflow and  freezes soon. 

You should change you TWBR =24 to =12 or substitute TWBR to Wire.setclock(400000),if only read data without mag ,mine is working fine.

batata004, 2016-07-11:
I only use MPU6050 without any magnetometer, and I try to read data from DMP (not raw data). It freezes after a while.

You think chaing TWBR from 24 to 12 should help?

yang614, 2016-07-11:
Yes, or Wire.setclock(400000).

batata004, 2016-07-11:
I tried it just right now. If freezed using "Wire.setClock(400000);" and also when using "TWBR = 12;". Any suggestion?

jaundice, 2016-07-11:
A while ago I had issues with FIFO overruns on a 9150 but I can't remember the details. Switching to FastWire fixed it for me.

batata004, 2016-07-11:
Sir, how I switch to fast wire?

TMcGahee, 2016-08-23:
This is not a bug - your loop is taking too long, and the buffer in the IMU is filling up and crashing the I2C bus/processor. That's why removing Serial prints and reducing delays (as you mention in your Arduino forum posts) solves the problem. That's why switching the code to check the FIFO will work. It manifests in many ways. Jeff has written repeatedly in his documentation to check the buffer as often as possible - adding a delay(100) is the opposite of that. In my experience, you need to check the buffer at least every 12ms to long term stability, but the faster the better. You should close this bug report.

batata004, 2016-08-23:
@TMcGahee you certainly have no idea what you are talking about and you surely should delete your reply to this bug report. Read all the thread, I am not using any delay and I know much better than you how important it is to keep the loop fast.

TMcGahee, 2016-08-23:
Wow. You're really just an angry brat, aren't you? No wonder no one is
trying to help you. You're wrong, but I'm not going to show you why. Learn
some etiquette

ewerlopes, 2016-10-11:
@batata004 , any progress on this matter?

batata004, 2016-10-11:
I gave up, this library is too buggy in many aspects and Jeff Rowberg does not care about solving these MANY bugs that you can find with a simple search at Google. Also Jeff does not even read these threads.

IMPORTANT: DONT USE JEFF'S LIBRARY FOR ANYTHING IMPORTANT. Sooner or later it will fail and if you are using this library with your quadcopter, robotic arm... STAY AWAY OF IT. You will get hurt and hurt other people. There are serious problems with this library which will, sooner or later, cause you intermitent problems.

jrowberg, 2016-10-11:
@batata004, please try to avoid sensationalist language with unsubstantiated claims. Yes, I am reasonably certain that the I2Cdev core and especially some of the device libraries are not bug-free. However, they have worked remarkably well with very few changes for literally thousands of people, some of whom have come back to tell me so after the fact (with complete, stable, working project demos).

As noted in my comment from June 28, I2Cdev core code itself does not contain any loops that could lock up the whole system. The Wire library, however, does. This is well-documented and tracked separately. It is not something that I2Cdevlib code can fix, other than switching to a different I2C implementation (which is not a bad idea, but it is outside the scope of this bug report). I still contend that I2Cdev core communication works correctly **as long as the hardware is electrically reliable**. If the hardware is not electrically reliable, the CPU freeze is because of the Wire library, which has a blocking `while()` loop and no timeout as documented in Arduino issue #1476.

I have taken the example code that you posted on June 28 and run it with no modifications in the following setup:

- Arduino IDE v1.6.11
- I2Cdevlib repo pulled from latest Github master branch
- Official Arduino Uno R3
- GY521 breakout board for MPU-6050
- AD0 pin tied to Arduino GND
- VCC pin to Arduino 5V (GY-521 has 3.3v regulator on board)
- SDA/SCL pins connected to Arduino SDA/SCL correctly
- Saleae Logic analyzer monitoring SDA and SCL

![Hardware setup](https://lh3.googleusercontent.com/-nSjcu3lPAupTVMRMKClzHkcdkVm_h6AUUlXl2wneu5Ue5cu9_75IxTCFbUZTLW7xP7X4BP7rV_cfS4BdbyYlkef8aKRyQui4BMCegxqmKaNxd45mh6p6ZiczqQIYXrCVgU7dvuo-N_zRObEw4onbvys2CkVWiqtNNGRnuJPGUV9JagL50HxJkDXTCkNQZd9Y8cYt7nS-phoOWmA3XJl6Ds9yKaqS2-TdIyzz4p1S1omzizHsWXrn45mCa7lc5UsUKek5M1tqJN4FQDGjWHQsa8uPwFqVmYvLH7QP3nT1GpKwK5Ze9XC8t_niDpJjprX3u2158JekPeTr8btmMtvTctNOF8kTr5bQm1EYxKnKT6VtJeUNnwYHOWknVJJ5svJLxwZ0OHmdv8tTeJr5tzxQqHPELBYIiPih0-u51DbUEYeSb9aYURgfVPEgQml_ROrV744IhELq7C61FKLKlIB6LBww76dcqv1DmnQDS0X1CNk70gc6fJ2ip5EThhRwiw_js_QVieQelYE1uXzXpv-d2uvOLpi4K6UpbShcRobzmqnr4yXTpIZf1iUgVkWKwSYf7vKr3JQ0gLB3uA_wNdFOCgnJILy3sizrqkCs_bRcXUau8Lv=w1027-h770-no)

Your sketch compiles and flashes correctly. The DMP initialization completes without issue, and the YPR output settles within a few seconds to a stable value. I have left it running for 15 minutes with no problems. It has remained totally stable with no freezes, FIFO overflows, or other problems.

To test, I pulled out the SCL wire to simulate an electrical issue. This (at least this time) caused the SDA line to get stuck low:

[image]

The I2Cdev core is not doing this inside its own abstraction code. The GY-521 has 2.2k pull-up resistors on the SDA/SCL lines, so something is driving SDA low during a *read* operation from the Arduino side. Reconnecting the SCL wire does not resume normal operation, even with all zero data coming back. This suggests that the Wire library got stuck in its blocking `while()` loop while the MPU-6050 sensor is holding SDA low waiting for more clock pulses, which won't come because the Wire library is stuck. (Maybe. I haven't dug into the Wire library's guts.)

But the main point here is that it worked perfectly until I broke the hardware on purpose.

I will reset the Arduino and let it run for another long while just to confirm this (expected) behavior.

batata004, 2016-10-11:
Thanks jeff and I didnt want to offend you. No. I appreciate your help, but I think it would be smart to let people know that your library sometimes crashes. Yes, I executed this same code many times with no problem. But sometimes when I just add some lines of code it gets completely unstable, and after a while it does not generate any other problem.

Jeff, I dont have bad wiring, this is not my case. The wires are very well connected to arduino.  I realized this bug happens when you call mpu.getFIFOCount() too many times in small time frame. It almost always freezes at mpu.getFIFOCount() (right when it gets executed). I know that for a fact cause I puted serial.println before and after that function, and it freezes the only the first serial.println.

You library may work fine in a moment, in another it just does not work. Adding some lines make the code break sometimes, and other times dont (there is no conflict between your code and those new lines).

Jeff, I appreciate your attention and efforts into this library but something is not right. Maybe you are right and it's bad wiring, but in my case it surelly is not. I tested this setup in other arduinos and got similar results.

I dont know if you ever heard of SEBRAE here in Brazil. It's an education institution "for free", with small fees. I know at least 2 good friends which are teachers of this institution in electronics and also confirmed the problem I was facing. It was them who made me come here and report this. 

I tried to report this problem in many different foruns and in almost all cases, in the middle of the thread, other people confirmed what I said and had the same problem with different setups.

If you think this problem happens of because of some lines of I2C getting LOW, why cant you setup and interrupt in your code to check if the line is LOW for too long and maybe reset the device it that timeout happens? I never used an original arduino, I always used clones, maybe some poor build in those clones cause this problem? 

Sorry if I disrespected you, you are a nice guy who devoted lots of time to this library however I think, IMHO, that it's not safe using your library in quadcopters, robotic arms... someone could get seriously hurt.

KyleMercer, 2016-10-20:
@batata004 I don't think you are quite getting the point Jeff is making. The problem is not with his library, but with the Arduino Wire library itself that is part of the Arduino IDE. More specifically it is the two-wire interface (TWI) library that the Wire library calls to handle I2C communications.

I stumbled upon this discussion looking for an alternative implementation of an I2C library for a unrelated device I am using and can confirm that **_any_** device that calls the Wire library will suffer the same issue you are talking about. 

While loops are used in the TWI library for communication, and as a result, the code can block all other functions on the Arduino while it is waiting infinitely in the TWI while loop. Jeff demonstrated the infinite loop by pulling out the SCL wire from his hardware. Internally, the TWI library provided by Arduino waits for a response from the sensor that is never coming because the physical wire is removed.

While your friends at SEBRAE confirmed the behavior, like you they have wrongly assumed it is due to Jeff's code. If you took a different I2C sensor with a library written by a different person (such as the one for the sensor I am using) and pulled a wire, it would behave the exact same way. 

As for why can't Jeff put an interrupt in his code to check if the line is LOW for too long, the library is blocking anything from happening (Which is why everything stops). The Arduino environment is not a Real-Time Operating System (RTOS). The issue is outside Jeff's code.

I hope this clarifies the issue.

batata004, 2016-10-20:
@KyleMercer you made it very clear, but it still does not explain why reading data from BMP180 using any library around will never hang, never. But as soon as I disconnect BMP180 and connect MPU6050 problems start to happen. It may be a problem in the WIRE implementation, ok! Maybe for some reason MPU6050 has some problem (I already tried at least 5 different modules from 3 different sellers in the last year or so) with the hardware, I dont know. I can only say that this problem only happens with me:

1) WHEN using MPU6050 with jeff's library
2) WEN I try to read veery fast data from MPU6050, if I read data once every 100ms, it all works fine
3) in unprecitable times, I cant be sure that it will happen now, but it will certainly happen at some moment

Anyway, you were all very clear and made a point. Maybe it's wire.h implementation that is faulty!

KyleMercer, 2016-10-21:
@batata004 Thanks for your response. I hope that no offence was taken. The issue I am having is actually related to the BME280, a distant relative of the BMP180 sensor, and communications via TCP/IP. My hardware and project is quite different from what you are discussing, and I do not use the MPU6050. From my quick Google search it sounds like the sensor does a lot, and making it work with other I2C hardware is tricky. I wanted to offer some feedback only because I noticed the strong language in your reply to Jeff.

I did a quick test with my project, in which the BME280 is the only device communicating via I2C. My microprocessor is an ESP8266 (Which is quite a bit faster than the Uno) and I am using a common industrial protocol over TCP/IP. I have some software that is polling the ESP8266 every second, and I am actively debugging using the serial interface, so there is additional work being done by the system. I pulled one of the wires, and the system kept working. I put the wire back, still no problems. I tried this a couple more times, randomly, and finally the system stopped working. I am only asking for one value via I2C and one value over the industrial communication channel, so I would guess that if I increased the amount of data through the different interfaces, the problem might start to happen more often.  

Anyways, I hope that you can find a solution to the problem you are having. I thought I would share what I have experienced. Take care!

batata004, 2016-10-21:
@KyleMercer  no offence taken :) Glad to know your case, indeed it looks even more a problem with I2C protocol itself than with the jeff's library. What you said is exactly what happens to me, randomly/intermitently problems (as you removed the wired a few times).

Anyway, I hope jeff or somebody here with great knowledge in this subject could file a feature request/bug report at Arduino community so they can bring some light to this and solve this problem.

salute2vishal, 2016-11-21:
@batata004 
Hey did you find the solution to the problem ?  I am also facing the same issue, sometimes the library gets stuck at fifo count function .

batata004, 2016-11-21:
@salute2vishal It's really sad but I got no solution for this problem. It already consumed hundreds of hours trying to work around this but sometimes I think I2C is the guilty but other times I thinks it's jeff's library the problem. It's really an intermitent problem and it's hard to make other people believe I am not doing something stupid - which I am not doing. I wish Jeff could take a look at this serious problem cause many many many other people are facing it on other forum posts and so far no solution.

jrowberg, 2016-11-21:
I have looked into the issue both with and without your supplied code, but have not been able to replicate it without intentionally severing the hardware I2C connection. Any additional troubleshooting will require a logic capture with a Saleae analyzer or similar during a moment when the problem actually occurs on your own platform. But even with this additional detail, I expect the root cause to be outside of any of the I2Cdev code.

batata004, 2016-11-21:
@jrowberg  thanks for helping. I dont have a logic analyzer, hope another person with this same problem that owns a logic analyzer can provide details about this problem.

piroozn, 2016-11-21:
batata004 
If you have a jtag emulator. It is the best way of finding issue. Since you are detecting the issue, it is best for you to debug it.
•	Just stop the processor at the failure point and see what is wrong.
•	Learn from each break point and set more break point to track the problem back to find the root cause.
Regards,
Pirooz

batata004, 2016-11-21:
@piroozn I never hard of "jtag emulator", sorry I dont have access to one, I will ask a good friend of mine tomorrow to know if he has one and can borrow me for a day.

piroozn, 2016-11-21:
I helped someone fix the issue with the same setup using the Arduino 101 libraries and fixed the performance issue in the same area that you are having problem.
Someone added a fixed delay after wire operation in the readbyte call that was slowing down the I2C accesses. Just remember if you use too many print statement in this area would create new issue or make the issue worse. If you print something, maybe you should print one character and only use one debug statement at a time until you get Jtag emulator.
Also remember you have multiple FIFO in the pipe and it may be getting confused dealing with its threshold setting. I2C controller FIFO, Accelerometer FIFO. Do both generate interrupt? If so disable one and use different method of hand shaking.

batata004, 2016-11-21:
@piroozn actually what you are saying is very true! If I dont call too frequently FIFO functions (or retrieve data) this problem does not happen, so your solution to add a delay will certainly help IF, AND ONLY IF, this delay is not too high which could cause the fifo overflow. 

This "bug" relies in the fact that if you request data too fast using I2C something will get wrong at some time and will freeze arduino. And what is really really strange is that sometimes when arduino freezes the watchdog fires, and sometimes not. I use watchdog to turn off my quad if arduino is not working for 500ms but sometimes arduino freezes and watchdog does not even fire.

piroozn, 2016-11-21:
@betata004 May be I did not say it correctly. The driver had not needed delay and I removed it to fix the issue.
If you have extra delay in the driver, when you call it then you have to wait the same amount of time or more otherwise thing get backed up and creates problem. 
It helps if you check status of thing before you write or read so no conflict will happen.

slesta, 2016-11-23:
@batata004 I have similar problems on Arduino nano. Arduino freeze after variable time and it depends on inv_set_fifo_rate.  It run cca 20 minutes when I set 0x04, but it is too slow for me. After reading this thread I started experimenting with TWBR. I use your sketch and now I have 100Hz fifo rate and TWBR 103, sketch is running more than 1 hour without problem. I don't know why, but may be it helps you.

ewerlopes, 2016-12-01:
Hi all, 

Well... I use Jeff's library together with the NRF24L01 Radio 2.4GHz transceiver library on an Arduino UNO. The idea is to read data from the MPU6050 and then transmit it for further processing by a nearby computer. My problem with that was that I would, like most of you, see the arduino freeze after some seconds. The chances of getting the problem was somehow correlated with the use of the serial print, but not just that in my case! 

I work on an university robotic lab, thus I have permission to use multiples arduino boards. One day, trying to debug the problem, I noticed that the code would run normally on some arduino board (not freeze) and not in others. I had those made in Italy and those made in China.. It turns out, only the Italian ones work.  If I upload the same code on the Chinese boards it doesn't work! So, I assume, there must be some hardware incompatibilities issues with the library. 

In summary, I removed all calls to serial print inside the arduino loop function and use only Italian made boards. In the receiver code (arduino connected to the computer), I use the jeff's library with ros, in order to read the transmitted data from the serial port. It works for me! The only thing now is that I want to receive data at 100Hz, now I receive at about 48Hz. Maybe, the @slesta tip on playing with TWBR may also work. Let's see!

ewerlopes, 2016-12-01:
Hi @jrowberg 

I need to get the raw values from the accelerometer normalized  - in the range [0,1], can you please give me some ideas on how to proceed? I saw you post [here](http://www.i2cdevlib.com/forums/topic/4-understanding-raw-values-of-accelerometer-and-gyrometer/) about the raw values received from the module.

> The output scale for any setting is [-32768, +32767] for each of the six axes. The default setting in the I2Cdevlib class is +/- 2g for the accel and +/- 250 deg/sec for the gyro.

Since I am using the default setting in the I2Cdevlib class, would it be correct to just divide the accelerometer x, y, z values by 32767? 

I appreciate your answer, thanks.

batata004, 2016-12-01:
Hi,

Indeed my boards are chinese or clone ones, I never had the chance to test this problem in a original board. It would be nice if someone could find a chinese board to Jeff so he could test this bug and maybe fix it.

@ewerlopes  if you want to "normalize" a value what you need to do is simple, is map function from arduino, see here https://www.arduino.cc/en/Reference/Map. Something like this:

float x = map(PLACE_YOUR_ACCEL_DATA_HERE,-32768,32768,0,1);

Hope it helps.

slesta, 2016-12-01:
Change TWBR on 103 realy help me, but it's not perfect. Now it's running few hours, but still freeze at the end. It's much better then few minutes or seconds. It all depend on amount of code and amount printed chars between fifo readings and on fifo rate. I have chinese clone too. I spent long time on searching for right value for TWBR. Results of different values are very variable from seconds to minutes. TWBR 103 runs in hours for me, you need another value, may be, you must try. I use 6050 2 years at chinese clones only and all that time I tried find the solution for this problem without succes. I will try buy italien version and test it, but better is find problem with chinese clones. Mr. Rowberg I buy one for you. I don't have skills to debug I2C lib.

batata004, 2016-12-01:
Indeed @slesta I already palyed with TWBR and it produces some different timings on the freeze, low values freezes faster. When I get home I will try 103 but the most time I could run the code without freezing is a few minutes, never got a full hour. Maybe your arduino clone is better than  the ones they sell here in brazil (also from china).

ewerlopes, 2016-12-01:
@batata004 I know how things are in Brazil regarding this. Boa sorte!

jrowberg, 2016-12-01:
@batata004, can you provide a link to an online store/auction (eBay, Aliexpress, etc.) for the same kind of clone that you have?

batata004, 2016-12-01:
@ewerlopes OBRIGADO! :)

@jrowberg the clone that I bought is from this seller:

http://produto.mercadolivre.com.br/MLB-723811978-arduino-atmega328-nano-30v-rev3-cabo-usb-robotica-automaco-_JM

And this:

http://produto.mercadolivre.com.br/MLB-705779338-arduino-nano-v3-_JM

I never bought it from eBay or Aliexpress.

piroozn, 2016-12-01:
@ ewerlopes
Keep in mind the FCC and Europe regulatory body require different power and transmission duty cycle that may be helping to show the issue more. Looks like we have throughput issue with something getting behind and overflowing a buffer or memory.
If you use Jtag emulator. You can stop the MCU at freeze point and see where we are. Then you can breakpoint at the start of that routine and see how you are getting there. The best tools for debugging this type of issue.

jrowberg, 2016-12-01:
This looks like the same thing:

http://www.ebay.com/itm/MINI-USB-Nano-V3-0-ATmega328P-CH340G-5V-16M-Micro-controller-board-Arduino-/201538986543?hash=item2eeca8e22f:g:Be4AAOSwp5JWbiLn

Close enough for testing?

batata004, 2016-12-01:
that looks really close, the only difference is that for some reason my arduino's driver is ch340 and this one looks like ch340g, I dont know if this makes any difference.

batata004, 2016-12-01:
Actually this one looks more close to mine -> http://www.ebay.com/itm/Nano-V3-0-Mini-USB-ATmega328-5V-16M-Micro-controller-Board-CH340G-F-Arduino-NEW-/162002876661?hash=item25b81fb8f5:g:ARcAAOSwuzRXefjh

The cheaper the better cause here in brazil they only sell garbage, so you should probably try to find the shittiest/cheapest arduino and try it out.

jrowberg, 2016-12-01:
I've ordered one of both. The one from China will take a lot longer to get here.

batata004, 2016-12-01:
Nice! I really hope you get a faulty one and you can check this problem so the community can benefit from your learnings :) Thanks man, I had already given up using your library, now I have high hopes I can come back to use it!

slesta, 2016-12-05:
I bought original Arduino nano on friday. Same code run till discharge of battery (cca 6hrs) on 100Hz.  @ewerlopes you are right. Thank you;) Mr. Rowberg thank's for your patience. I hope you find solution of clone troubles. And thank's for your libraries.

ewerlopes, 2016-12-05:
@slesta I am happy to see that the problem is somehow solved! By the way, I tried to modify the TWBR  value but it did not work. I keep receiving data at about 48Hz. However, this sample frequency is not a huge problem to me now. Good luck!

slesta, 2016-12-05:
@ewerlopes  TWBR is I2C rate, look here http://www.gammon.com.au/forum/?id=10896#i2csummary . If you want change dmp rate, you must change it in library. 
Search for this part in 6Axis_Motion and change AxAA to desired value.  It's near line number 300.

`0x02,   0x16,   0x02,   0x00, AxAA                // D_0_22 inv_set_fifo_rate 0x03 - posledni

    // This very last 0x01 WAS a 0x09, which drops the FIFO rate down to 20 Hz. 0x07 is 25 Hz,
    // 0x01 is 100Hz. Going faster than 100Hz (0x00=200Hz) tends to result in very noisy data.
    // DMP output frequency is calculated easily using this equation: (200Hz / (1 + value))`

I tried 200Hz (0x00) and it's run 200Hz and it's independent on TWBR. I change TWBR for better stability of chinese clone only, not for change of DMP speed.

catgirldo, 2016-12-09:
Sorry to be late to the party. I had struggled with what I believe was the same problem; the program would run and then randomly lock up. I have to echo what slesta came up with. In the file "MPU_6050_6Axis_MotionApps20.h" down at line 266.  I changed the 0x01 which is 100HZ to 0x02 which is about 66HZ and I get a pretty reliable readout. I have yet to test it on a mobile platform, but it is pretty funny to have it in the car with me and watch it give me yaw information when I turn.  My set up is an Ardunino Nano V3.1 with a Level Translator Breakout - PCA9306 to interface with the MPU.  Yaw drifts about 1 degree every minute and a half or so but for my application this should work. Hope this helps.

batata004, 2016-12-09:
@catgirldo to me it does not matter the rate I read the buffer, eventually it will freeze. Could you please confirm that you arduino is original or a clone?

You yaw drift I think is expected, you could use a magnetometer to fix that, nothing too complicated.

catgirldo, 2016-12-10:
batata, not sure what you mean by original or clone. The box says it is " authentic official Arduino" made in the USA by Gravitech so I'd say it's an original but not made in Itay. I switched it on yesterday and it was still running after 7 hours.  I also ran the code on an obsolete product that was Atmel based and had the same problem with freezing. I have not gone back to see if I could solve the problem with a code change but that was the reason I started using the Arduino in my project. I thought there was a bug in the other processor but the Arduino initially experienced the same issue. It is possible the clone has some "overhead" issue but my experience with the Atmel makes me think its a generic problem. In any event, if I run the nano at 100hz it will freeze up randomly. Speed seems to be the issue so one possibility is the oscillator  on the Arduino board. At this point, I have built two units and both operate in an identical fashion and since someone else runs a nano at 100hz, I'd suspect a tolerance issue in the oscillators but that is just a guess. Good luck!

batata004, 2016-12-10:
@catgirldo Your comment is of great value, mainly for Jeff. IMHO your arduino is probably original cause it's working fine (as you reported). I cant know for sure if this problem is related to some overhead in clone arduinos, but surelly jeff will get an answer as soon as his clone arduino arrive from china. 

Speed was surelly related to the bug, the faster the read from the buffer more frequently arduino freezes, but in my case, no matter which speed I choose, the bug always happens in a 10min span time.

Thank you for your explanation, hope it helps some one more in depth with the "inner workings" of I2C and Atmel processors.

[26 more replies]