jrowberg/i2cdevlib #261: MPU6050 on Leonardo requires different I2C and interrupt pins

MauiJerry, 2016-07-29:
Reference the DIY Hacking article and my new comment on it...
  http://diyhacking.com/arduino-mpu-6050-imu-sensor-tutorial/#comment-3543

The article does a nice job showing hookup an arduino Uno... but we had a Leonardo (and some others).  The Leonardo puts its I2C on pins 2(SDA) and 3(SCL) which precludes using D2 for interrupt.   A bit of googling and we found the solutions...

Up at the top, where mpu is defined, we add the lines:
```c
// The SDA/SCL are on A4/A5 on normal arduinos
// on Leonardo 2 (SDA), 3 (SCL)
// see Wire library reference   https://www.arduino.cc/en/Reference/Wire
// also http://arduino.stackexchange.com/questions/4019/how-to-connect-the-int-pin-of-a-mpu-6050
// Also on Leonardo connect MPU6050 interrupt pin 7,  Otherwise it is on D2
// use the function to convert pin# to interrupt id
//#define MPU6050_INT_PIN 2
#define MPU6050_INT_PIN 7
#define MPU6050_INT digitalPinToInterrupt(MPU6050_INT_PIN)
```

Then down in Setup, we changed the attachInterrupt section to...
```c
        // enable Arduino interrupt detection
        Serial.print("Enabling interrupt detection (Arduino external interrupt "); 
          Serial.print(MPU6050_INT);Serial.print(" on pin ");Serial.print(MPU6050_INT_PIN);Serial.println("...");
        pinMode(MPU6050_INT, INPUT);
        attachInterrupt(MPU6050_INT, dmpDataReady, RISING);
```
and it all works fine on both arduino and other board.
Hope the formatting works.

BasheerSience, 2017-03-26:
That's awesome @MauiJerry you saved my life 👍

kalenykkonstantyn, 2017-04-03:
Only in this video i found the right answer for our problems (arduino Leonardo + mpu 6050) [videoI2c](https://www.youtube.com/watch?v=xU_mmpvkXEg)

jxy4769, 2020-09-11:
Thank you so much! I was searching the internet for literally hours and this was the best solution.