jrowberg/i2cdevlib #265: Wrong use of Wire library

Koepel, 2016-08-20:
A few issues have already been reported about code with the Wire functions of I2Cdev.cpp.
https://github.com/jrowberg/i2cdevlib/issues/252  (timeout is not in Arduino library)
https://github.com/jrowberg/i2cdevlib/issues/181 (adding a repeated start)
https://github.com/jrowberg/i2cdevlib/issues/76 (data length)

In 5 places, the Wire.requestFrom() is followed by a Wire.endTransmission(). That causes an extra I2C transaction with the address on the I2C bus and the Slave acknowledging. The Wire.requestFrom() is a complete I2C transaction on its own, and should not be followed by a Wire.endTransmission().

A timeout for getting data after Wire.requestFrom() is not needed. When the Wire.requestFrom() returns, the I2C transaction has completely finished and the received data is in the receive buffer in the Wire library. That data can be read with Wire.read(). In case there was a bus error or a collision, the return value of Wire.requestFrom() or Wire.available() will be less than the number of requested byte (it will most likely be zero). A single check for that error or collision is possible.

reloxx13, 2018-04-14:
@jrowberg  old but missing feedback :(

ONLYstcm, 2020-05-24:
@Koepel why not make a pull request? I've noticed you been bringing this up on several MPU6050 libraries and I've actually made the necessary changes on my end but I feel like this might go over alot of developers heads especially if they encounter errors or weird behaviour.

Koepel, 2020-05-24:
@ONLYstcm There is too much going on. Without the bugs the FastWire can be removed. The Arduino Wire library does not have timeouts so all the timeouts can be removed. But then there is a Teensy I2C library that does have timeouts. For that Teensy library it would be nice to keep the timeouts. Then there is a NBWIRE which uses some of the source of the Arduino Wire library. But I don't know if bugfixes from the Arduino library made it into NBWIRE. The FastWire and NBWIRE are only for AVR chips. 
ShallI go on ?

It all started with the Arduino Wire library which does not have timeouts. The library became a standard from the start, and everything went haywire (pun intended). They could have fixed that problem for the SAMD processors, but also that Arduino Wire library does not have timeouts.

The Arduino [reference for the Wire library](https://www.arduino.cc/en/reference/wire) was not good. That is improved, and it continues to improve over the years with very small steps. My [alternative explanation](https://github.com/Koepel/How-to-use-the-Arduino-Wire-library/wiki/Explanation-of-the-functions-of-the-Wire-library) still makes a few things more clear, which should already be clear by reading the reference.

ONLYstcm, 2020-05-24:
@Koepel I've already read your wiki and actually used that to make necessary changes on one of the libraries that I was using. But yeah please do go on, this will be a good reference point for others as well.