jrowberg/i2cdevlib #266: RA, TC ... What do they stand for?

streamnsight, 2016-09-06:
What do RA, and TC prefix stand for? 
I couldn't figure this from the docs.

Thanks.

jrowberg, 2016-09-08:
The "RA" segment of a constant typically means **register address**, while everything after that comes from the register map or datasheet and is defined by the manufacturer. In the case of "MPU6050_RA_XG_OFFS_TC" and similar, I believe "TC" stands for test constant or test calibration or something similar. The register map is not clear about those registers and what they do.

closed 2016-09-08 by jrowberg

streamnsight, 2016-09-08:
Thanks! 
it makes more sense now.

streamnsight, 2016-09-08:
by the way, are you familiar with the fact that InvenSense published the actual code for the MPU6050 DMP functions?

This includes quite a few interesting features such as tap detection, fall detection, no motion detection, heading, orientation detection etc...

the latest can be download in their Developper section (registration required) under Motion Driver 6.2