jrowberg/i2cdevlib #269: [RaspberryPi] wrong results from readBit (I2Cdev)

createproblem, 2016-10-06:
Hello I've some problems with this lib for the RaspberryPi. This function https://github.com/jrowberg/i2cdevlib/blob/master/RaspberryPi_bcm2835/I2Cdev/I2Cdev.cpp#L64-L70 gave me wrong results.

If I read 1byte from the register the correct result is: `0100 0000` I2Cdev lib did everything correct.
Now I'want to read the bit number 6 (second number from left) with `I2Cdev::readBit(dev, reg, 6, &buffer)`. The function gave me the result `0000 0000` That is not correct my expected result is `0000 0001`.

I changed the `I2Cdev::readBit()` function to the following:

Solution 1
----------
```cpp
int8_t I2Cdev::readBit(uint8_t devAddr, uint8_t regAddr, uint8_t bitNum, uint8_t *data) {
    return readBits(devAddr, regAddr, bitNum, 1, data);
}
```
This fix gave me the correct result `0000 0001` as expected. I checked the function again to understand whats going on.

In Line 68: https://github.com/jrowberg/i2cdevlib/blob/master/RaspberryPi_bcm2835/I2Cdev/I2Cdev.cpp#L68
```cpp
*data = recvBuf[1] & (1 << bitNum);
```
That look not correct for me why is the data in ```recvBuf[1]``` instead ```recvBuf[0]```
And I also miss the right shift line by bitNum ```*data >>= bitNum```. So I changed the function to the following:

Solution 2
----------
```cpp
int8_t I2Cdev::readBit(uint8_t devAddr, uint8_t regAddr, uint8_t bitNum, uint8_t *data) {
  bcm2835_i2c_setSlaveAddress(devAddr);
  sendBuf[0] = regAddr;
  uint8_t response = bcm2835_i2c_write_read_rs(sendBuf, 1, recvBuf, 1);
  *data = recvBuf[0] & (1 << bitNum);
  *data >>= bitNum;
  return response == BCM2835_I2C_REASON_OK ;
}
```

Now the function works as expected for me. 
Can someone confirm that behavior? Maybe I'm using the function in a wrong way?

And why is the function ```I2Cdev::readBit()``` not using ```I2CDev::readBits()``` to stay in the same logic and reduce duplicate code.

Thanks for Help.

createproblem, 2016-11-05:
No support here? @jrowberg

jrowberg, 2016-11-06:
@createproblem The RasPi variant of the library was contributed by others; I haven't used I2C devices on a RasPi before. My guess is that the `recvBuf[1]` index is an off-by-one error.

Looking back at my original (Arduino) code here, there is some inconsistency between `readBit` and `readBits`, since `readBit` does not right-shift the result into the Bit 0 position. This is not a problem if you are checking for zero vs. non-zero results, but it can be a problem if you are checking for `1` explicitly, as this will only work when reading Bit 0. But I suspect this is not related to the problem you are observing.