jrowberg/i2cdevlib #271: Issue compiling on the Arduino Zero

mundodisco8, 2016-10-18:
Hi!

I'm facing some problems compiling some code on my Arduino Zero. I run the latest versions (IDE v1.6.12, SAMD boards v.1.6.8). I also just cloned the repo, so I'm running the latest version of I2Cdevlib.

When I try to compile my code, it returns the following error:

```
D:\Joel\Desktop\Dropbox\Arduino\libraries\I2Cdev\I2Cdev.cpp: In static member function 'static int8_t I2Cdev::readBytes(uint8_t, uint8_t, uint8_t, uint8_t*, uint16_t)':

D:\Joel\Desktop\Dropbox\Arduino\libraries\I2Cdev\I2Cdev.cpp:276:62: error: 'BUFFER_LENGTH' was not declared in this scope

             for (uint8_t k = 0; k < length; k += min(length, BUFFER_LENGTH)) {

                                                              ^

C:\Users\Joel\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.7\cores\arduino/Arduino.h:96:24: note: in definition of macro 'min'

 #define min(a,b) ((a)<(b)?(a):(b))
```

I checked Wire.h. In the SAMD version, it lacks a BUFFER_LENGTH constant that is present in the AVR version of the library. Consecuently, the code compiles for an arduino Uno, for example, as BUFFER_LENGTH is defined here.

I found a related issue (link to a possible solution)

https://community.particle.io/t/wire-library-error-porting-mpu6050-driver-with-related-library-solved/3213/102

Where they modify I2Cdev.h by adding the variable, with the value of BUFFER_LENGTH of 32, saying that is the MPU6050 buffer length. I don't get that as my understanding is that the buffer is 1024 bytes.

Is this a bug? Or I am using the library in an unsupported mode by using it on the arduino Zero, being this library targeted only to AVR controllers?

Just including the library is enough to trigger the error:
```
#include<Wire.h>
#include "MPU6050_6Axis_MotionApps20.h" // for the MPU6050

void setup(){

}
void loop(){

}
```

mundodisco8, 2016-10-21:
I got an answer from the guys at Arduino at the referenced ticket 

https://github.com/arduino/ArduinoCore-samd/issues/175#issuecomment-255111896

jensykesalloneword, 2016-10-25:
Hello, 
what method did you use to resolve this? Is it an edit to the i2cdev.h file or mpu6050.h file? I'm having a similar issue with an Adfruit feather m0.
From reading your other reference ticket and the thread about the particle.io am i right in thinking it stemmed from an update to the SAMD in Arduino? 
I have tried to increase the buffer size in the i2cdev.h but this hasn't resolved the issue. 
Thanks

jensykesalloneword, 2016-10-25:
Please ignore I now have it resolved I was defining it incorrectly. Thanks for the useful thread !

mundodisco8, 2016-10-26:
I see that you solved it, but in case someone in the future stumbles upon this, what I did was to add 
`#define BUFFER_LENGTH 32`

To my wire.h file, which is located in 
`C:\Users\YOURUSERNAME\AppData\Local\Arduino15\packages\arduino\hardware\samd\YOURBOARDVERSION\libraries\Wire`

Be careful to change the right Wire.h, as you have one for each of the architectures you added to your arduino IDE installation.

I haven't tested my code. I DON'T KNOW IF THIS CHANGE WILL BREAK SOMETHING DOWN THE LINE. From the developer's answer, we can see that the Zero uses a different buffer structure than the AVR family, so it might work or it might not. But at least it compiles now.

HamAndEggs, 2017-04-12:
I have just ht this error. I have an arduino zero ide version 1.8.1 and I downloaded your libs 11/04/2017 (uk date format)

Same error as above. Seem two suggested fixes, with their BUFFER_LENGTH defined as 32 or 64 bytes. Does it matter which I used????

Thanks.

mundodisco8, 2017-04-12:
I ended up adding
`#define BUFFER_LENGTH 32`

To my I2CDev.h file (instead of to my wire.h), as I think that if wire.h removed that variable, then it doesn't belong there, but it's just my opinion. I choosed 32 because that was the value before I updated my boards in the Arduino IDE. My code seem to work fine, but it's only circumstantial, as I haven't run tests to see if that breaks something down the line. You will need to run your own tests, I'm afraid :S

amadeok, 2018-09-30:
Hi, i tried compiling for an Itsy Bitsy m4 and got the same error, and you fix solves it. Have you run into any issues? thanks

amadeok, 2018-09-30:
> I ended up adding
> `#define BUFFER_LENGTH 32`
>
> To my I2CDev.h file (instead of to my wire.h), as I think that if wire.h removed that variable, then it doesn't belong there, but it's just my opinion. [...]

Hi, i tried compiling for an Itsy Bitsy m4 and got the same error, and you fix solves it. Have you run into any issues? thanks

mundodisco8, 2018-10-01:
I never tested it properly, but didn't hear about any errors from the
colleague that did the FW.

closed 2021-09-26 by jrowberg