jrowberg/i2cdevlib #283: MPU9150 DMP Quaternion :: yaw always starts at 0 (mag offset init in dmpInitialize ?)

benkuper, 2016-12-22:
Hello,
i'm using the lib to get basic Quaternion orientation, everything works, the quaternion outputs correct data (with some minimal yaw drift at start that is corrected within 10s max).

But i noticed that each time at power down the device and power up again, no matter the orientation of the device, the yaw will show 0 for the orientation from which the device has been turned on. Then it will output correctly the yaw depending on this first initialization (meaning if i turned 90° left, it will show -90).

So my question is : how to get an "absolute" yaw value, that i only need to offset once, and then when i power down and power up again the device, it would show the same yaw value for the same device orientation ?

Hope i'm clear enough.
Thanks for this wonderful lib !

FreakyMex, 2017-08-29:
Did you find a solution ?

otuaku95, 2018-05-24:
That is exactly the code that I need!
could you please send it to me because jrowberg library is giving me a random yaw initial value.
Thanks!

benkuper, 2018-05-24:
sorry otuaku, i don't have this code anymore... you might find what you need here, as it is using the MPU9250 in the same way : https://github.com/benkuper/BenTo-Legacy/tree/master/firmware/arduino/BentoV4