jrowberg/i2cdevlib #284: MPU6050 DMP - Reading the whole packet

Johnny1010, 2016-12-30:
Hi everyone,

I am running the DMP6 example sketch successfully to read yaw, pitch and roll (https://github.com/jrowberg/i2cdevlib/blob/master/Arduino/MPU6050/examples/MPU6050_DMP6/MPU6050_DMP6.ino).

I was wondering that in the following code where a 42 byte packet is read:

`       else if (mpuIntStatus & 0x02) 
        {
        // wait for correct available data length, should be a VERY short wait
        while (fifoCount < packetSize) fifoCount = mpu.getFIFOCount();`

        // read a packet from FIFO
        mpu.getFIFOBytes(fifoBuffer, packetSize);
        
        // track FIFO count here in case there is > 1 packet available
        // (this lets us immediately read more without waiting for an interrupt)
        fifoCount -= packetSize;`

Is it possible to replace this line 

mpu.getFIFOBytes(fifoBuffer, packetSize);

with this

mpu.getFIFOBytes(fifoBuffer, packetSize-28);

Because the quaternions are only present in the first 14 bytes which are being used to calculate ypr. Rest contains the raw accel and raw gyro readings which I don't need.

Thanks in advance.

jrowberg, 2017-01-01:
Reading only 14 bytes won't work if the DMP is still generating 42-byte packets, since the remaining data will remain in the FIFO buffer and the next 14-byte read will get something which is not DMP data. You have to read the entire packet out each time, even if you ignore some of the data.

What you actually want to do is reconfigure the DMP such that it doesn't put anything except the DMP quaternion into the output stream. However, I am unsure how to do this (although I believe it's technically possible).

Johnny1010, 2017-01-01:
Hi Jeff,

Thanks for the reply. I might have found a way around this, I am reading only the 14 bytes and then I reset the FIFO this way the new data that appears is always the correct 42-byte DMP data. This isn't of course the best way but seems to be working so far. Reason for reading just the 14 bytes is of course to save the time.

I just had another query, that is it possible to rearrange the FIFO data like if I want to swap the gyro readings in the FIFO with the acceleration data. Because I don't want the gyro readings, I only need the accel and quaternions so if they could both be present in the first 26 bytes so I could just read that.