jrowberg/i2cdevlib #288: MPU-6050 DMP FIFO read from an ISR

mizamae, 2017-01-23:
Hello,

I wonder if it would be possible to read the data from the FIFO of an MPU-6050 directly from an ISR.

I tried it somehow but nothing is retrieved and the Arduino hangs.

has anyone done something similar?? I do not even know if it is possible as it would need to communicate with the I2C bus from the ISR and not sure if its possible (interrupt inside an interrupt??)

thanks a lot!!

jrowberg, 2017-01-23:
While there _might_ be rare corner cases where this works under exactly the right circumstances, it won't normally work and is generally a bad idea. ISRs should be as short as possible, with any "heavy lifting" done outside of the ISR. Your application should use the ISR to set a "read the FIFO" flag, then have the main program loop check for that flag and act accordingly.

piroozn, 2017-01-23:
Your software architecture can be designed to allow that if OS and all users know we are taking longer in the ISR. In general that is not a good idea but if it is designed in then is ok. Take a look at what the Zephyr OS does. IT has fibers running actual state machine code in the ISR.
Just disable the interrupt and re-enable it when you exist. The pending interrupt will interrupt you when you exit.
You are correct and it is better to handle this outside of the interrupt handler. You should set the Fifo threshold to interrupt at higher number of Fifo entry to generate the interrupt and handle groups of entry to make it more efficient. Of course your application requirement tell you if you can do this.

PaulStoffregen, 2017-01-25:
This might be problematic on some platforms, particularly AVR-based Arduino where the Wire library doesn't work at all if used from ISR context.

closed 2018-01-17 by mizamae