jrowberg/i2cdevlib #289: Sudden angle jump read

mizamae, 2017-02-01:
Hello,

I am using an Arduino to read the YPR angles from an MPU6050 using the DMP feature. I log the angles prefectly but suddenly, some samples fail leading to a Roll angle of 90º even with the sensors laying on the table. Han anyone experienced anything similar?? does anybody know a solution for this (maybe some configuration??). I am using The standard library with FIFO freq of 20 Hz and the DLPF at 42 Hz (have tried changing these values and still happens).

Thanks!! (atrtached a file with the exported angle values where these events are shown)
[Log_values.xlsx](https://github.com/jrowberg/i2cdevlib/files/744126/Log_values.xlsx)

thomasck, 2017-02-06:
What hexadecimal value are you using to set the FIFO rate in the dmpconfig array?  Also make sure that the setRate() function in dmpInitialize sets the same rate. For example I am using 0x01 - 100hz in the config array and setRate(9) 1khz / (1+9) = 100hz in the dmpInitialize function. I have found that if these two rates are different there can be angle errors.

piroozn, 2017-02-06:
@thomasck
Don't forget that you have another level of fifo in the interface section (I2C, SPI, …). Your software changes or architecture needs to maintain the correct fifo level to generate interrupt and handler that can empty it before new data is coming in from the sensor.

thomasck, 2017-02-06:
@piroozn I think Jeff's example code for the mpu6050 handles this pretty well, assuming the code runs fast enough on the hardware. Arduino should definitely be able to handle 20hz fifo update.

closed 2018-01-17 by mizamae