jrowberg/i2cdevlib #291: Possible DMP latency?

thomasck, 2017-02-07:


piroozn, 2017-02-07:
@thomasck
To get the information on position changes ASAP. You need to configure all the Fifo's used to communicate this information to the CPU to interrupt after 1 entry and not multiple. Sensor device, SPI or I2C or UART Fifo both on the sensor and the MCU side.

thomasck, 2017-02-08:
@piroozn thanks for the reply! 
Are you familiar with the code in MPU9150_9Axis_MotionApps41.h that is used to set up the dmp and read Fifo data? The dmp data in the Fifo _should_ be updating at a fixed rate: whatever rate is set in line 305 of MPU9150_9Axis_MotionApps41.h, which allows up to 200hz. However for some reason the rate is also set in line 438; the setRate() functions sets the SMPRT_DIV (0x19) byte which governs sample rate for everything based on one of the gyroscopes' output rate. The example code provided by Jeff Rowberg then ensures the Fifo is read whenever the fifo buffer fills up with 48 bytes.

I believe this is the correct way to read the data for my purposes, since ideally I want all feedback (orientation, rotation rate, and acceleration) at the same time. However the results I am getting during flight imply that some of the data may have some latency.

Are you suggesting that I abandon trying to use the fixed rate feature of the dmp Fifo, and just read the newest data in the Fifo immediately when it is available? 

I apologize if you understand all of this and I have completely missed what you are trying to tell me.

-Thomas

piroozn, 2017-02-08:
@thomasck
I am not familiar with MPU9150_9Axis but have used others that are similar and have seen the same issue repeating.
The issue is that most of the libraries trying to make it easy for people to develop product without understanding the architecture and just by using API's and libraries  providing some example code.
This method usually generate just functional product and not a high performance one. To fix this you need to throw expertise at it or do some study on the Sensor and the MCU architecture document (Big picture). Please take a look at the datasheet for all parts and look at block diagrams and see where you have fifo's and how the 200 hz is generated. You will see they have different setting for the rate than adjusting the Fifo level initiating interrupt. These fifo's exist in most of the interfaces also.
If you want quick response to position changes. Set all fifo's threshold level to generate interrupt at 0 or 1 which ever generate interrupt with a single entry into the fifo.
When you receive the interrupt, provide the main action that needs to be taken in the Interrupt routine and rest of the lower priority part of it outside the interrupt. Most of the small MCU's in the market are using event driven OS and most people misunderstand how that works. You need to know the basic for this type of OS if you are using it. Without basic understanding of that, you won't be able work around its restrictions and take advantage of its benefits.
•	Understand your OS
•	Understand your devices architecture and system architecture
•	Write your software and validate each section to make sure it does what you intended. Use software (debuggers, IDE's like eclipse) and hardware tools (Jtag emulator) to validate your configuration of interfaces and sensors.
Regards,
Pirooz

thomasck, 2017-02-09:
Now that you mention it I have noticed a few things that aren't quite right with this MPU9150 library; for instance changing the full scale range of the gyro seems to have some major effects other than sensitivity. I think this may have been part of my problem.

The hard part with this particular sensor (MPU 6 or 9 dof chips) is that Invensense has released very minimal documentation about their digital motion processor (DMP). The data sheets for the chips themselves are available, and explain how to get raw data, but they leave out all information about the DMP and how to use it. 

Thanks for explaining this to me though, I will study up on this chip.

Cheers,

Thomas