jrowberg/i2cdevlib #307: Detailed MPU 6050 Calibration?

VytautasDV, 2017-05-15:
I dont understand how to use data of calibration, which values should i put in offset?

This is my last line before done line.

[-2844,-2844] --> [0,9]	[-2354,-2354] --> [0,6]	[1227,1227] --> [16381,16385]	[10,11] --> [0,4]	[-24,-23] --> [-2,1]	[-31,-31] --> [0,1]
-------------- done --------------

what do these numbers mean?
[-2844,-2844] --> [0,9].
to my understanding this [-2844,-2844] is offset values, but what are these [0,9]?

these are gyros offset data
[10,11] --> [0,4];      [-24,-23] --> [-2,1];      [-31,-31] --> [0,1]

which ones to use?

Please give detailed info.

alexzk1, 2017-05-18:
https://github.com/alexzk1/startrack/tree/master/arduino_libs/my_updates_to_others/calibrate_each_mpu6050

VytautasDV, 2017-05-21:
I had similar code, but it was stuck on endless loop and after 2 hours of running the code i could not get the offsets. But this one is quite similar and it worked, thanks.

closed 2017-05-21 by VytautasDV