jrowberg/i2cdevlib #309: MPU 6050 compiling problem with Arduino MEGA 2560

mateush888, 2017-05-22:
The code won't compile, I was copying files from the browser, but later I downloaded the .zip and did all again. I still get these messages:

In file included from Measure_angle_Arduino_MPU6050.ino:4:0:
C:\Users\Mateusz\Documents\Arduino\libraries\MPU6050/helper_3dmath.h: In member function 'float Quaternion::getMagnitude()':
C:\Users\Mateusz\Documents\Arduino\libraries\MPU6050/helper_3dmath.h:74:46: error: 'sqrt' was not declared in this scope
             return sqrt(w*w + x*x + y*y + z*z);
                                              ^
C:\Users\Mateusz\Documents\Arduino\libraries\MPU6050/helper_3dmath.h: In member function 'float VectorInt16::getMagnitude()':
C:\Users\Mateusz\Documents\Arduino\libraries\MPU6050/helper_3dmath.h:111:40: error: 'sqrt' was not declared in this scope
             return sqrt(x*x + y*y + z*z);
                                        ^
In file included from Measure_angle_Arduino_MPU6050.ino:4:0:
C:\Users\Mateusz\Documents\Arduino\libraries\MPU6050/helper_3dmath.h: In member function 'float VectorFloat::getMagnitude()':
C:\Users\Mateusz\Documents\Arduino\libraries\MPU6050/helper_3dmath.h:178:40: error: 'sqrt' was not declared in this scope
             return sqrt(x*x + y*y + z*z);
                                        ^
In file included from Measure_angle_Arduino_MPU6050.ino:6:0:
C:\Users\Mateusz\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h: At global scope:
C:\Users\Mateusz\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:325:32: error: no 'uint8_t MPU6050::dmpInitialize()' member function declared in class 'MPU6050'
 uint8_t MPU6050::dmpInitialize() {
                                ^
C:\Users\Mateusz\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:548:34: error: no 'bool MPU6050::dmpPacketAvailable()' member function declared in class 'MPU6050'
 bool MPU6050::dmpPacketAvailable() {
                                  ^
C:\Users\Mateusz\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:575:66: error: no 'uint8_t MPU6050::dmpGetAccel(int32_t*, const uint8_t*)' member function declared in class 'MPU6050'
 uint8_t MPU6050::dmpGetAccel(int32_t *data, const uint8_t* packet) {
                                                                  ^
C:\Users\Mateusz\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:583:66: error: no 'uint8_t MPU6050::dmpGetAccel(int16_t*, const uint8_t*)' member function declared in class 'MPU6050'
 uint8_t MPU6050::dmpGetAccel(int16_t *data, const uint8_t* packet) {
                                                                  ^
C:\Users\Mateusz\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:591:67: error: no 'uint8_t MPU6050::dmpGetAccel(VectorInt16*, const uint8_t*)' member function declared in class 'MPU6050'
 uint8_t MPU6050::dmpGetAccel(VectorInt16 *v, const uint8_t* packet) {
                                                                   ^
C:\Users\Mateusz\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:599:71: error: no 'uint8_t MPU6050::dmpGetQuaternion(int32_t*, const uint8_t*)' member function declared in class 'MPU6050'
 uint8_t MPU6050::dmpGetQuaternion(int32_t *data, const uint8_t* packet) {
                                                                       ^
C:\Users\Mateusz\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:608:71: error: no 'uint8_t MPU6050::dmpGetQuaternion(int16_t*, const uint8_t*)' member function declared in class 'MPU6050'
 uint8_t MPU6050::dmpGetQuaternion(int16_t *data, const uint8_t* packet) {
                                                                       ^
C:\Users\Mateusz\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:617:71: error: no 'uint8_t MPU6050::dmpGetQuaternion(Quaternion*, const uint8_t*)' member function declared in class 'MPU6050'
 uint8_t MPU6050::dmpGetQuaternion(Quaternion *q, const uint8_t* packet) {
                                                                       ^
C:\Users\Mateusz\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:632:65: error: no 'uint8_t MPU6050::dmpGetGyro(int32_t*, const uint8_t*)' member function declared in class 'MPU6050'
 uint8_t MPU6050::dmpGetGyro(int32_t *data, const uint8_t* packet) {
                                                                 ^
C:\Users\Mateusz\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:640:65: error: no 'uint8_t MPU6050::dmpGetGyro(int16_t*, const uint8_t*)' member function declared in class 'MPU6050'
 uint8_t MPU6050::dmpGetGyro(int16_t *data, const uint8_t* packet) {
                                                                 ^
C:\Users\Mateusz\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:648:66: error: no 'uint8_t MPU6050::dmpGetGyro(VectorInt16*, const uint8_t*)' member function declared in class 'MPU6050'
 uint8_t MPU6050::dmpGetGyro(VectorInt16 *v, const uint8_t* packet) {
                                                                  ^
C:\Users\Mateusz\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:658:91: error: no 'uint8_t MPU6050::dmpGetLinearAccel(VectorInt16*, VectorInt16*, VectorFloat*)' member function declared in class 'MPU6050'
 uint8_t MPU6050::dmpGetLinearAccel(VectorInt16 *v, VectorInt16 *vRaw, VectorFloat *gravity) {
                                                                                           ^
C:\Users\Mateusz\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:666:92: error: no 'uint8_t MPU6050::dmpGetLinearAccelInWorld(VectorInt16*, VectorInt16*, Quaternion*)' member function declared in class 'MPU6050'
 uint8_t MPU6050::dmpGetLinearAccelInWorld(VectorInt16 *v, VectorInt16 *vReal, Quaternion *q) {
                                                                                            ^
C:\Users\Mateusz\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:678:61: error: no 'uint8_t MPU6050::dmpGetGravity(VectorFloat*, Quaternion*)' member function declared in class 'MPU6050'
 uint8_t MPU6050::dmpGetGravity(VectorFloat *v, Quaternion *q) {
                                                             ^
C:\Users\Mateusz\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:689:56: error: no 'uint8_t MPU6050::dmpGetEuler(float*, Quaternion*)' member function declared in class 'MPU6050'
 uint8_t MPU6050::dmpGetEuler(float *data, Quaternion *q) {
                                                        ^
C:\Users\Mateusz\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:695:85: error: no 'uint8_t MPU6050::dmpGetYawPitchRoll(float*, Quaternion*, VectorFloat*)' member function declared in class 'MPU6050'
 uint8_t MPU6050::dmpGetYawPitchRoll(float *data, Quaternion *q, VectorFloat *gravity) {
                                                                                     ^
C:\Users\Mateusz\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:708:67: error: no 'uint8_t MPU6050::dmpProcessFIFOPacket(const unsigned char*)' member function declared in class 'MPU6050'
 uint8_t MPU6050::dmpProcessFIFOPacket(const unsigned char *dmpData) {
                                                                   ^
C:\Users\Mateusz\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:718:84: error: no 'uint8_t MPU6050::dmpReadAndProcessFIFOPacket(uint8_t, uint8_t*)' member function declared in class 'MPU6050'
[... 75 lines cut ...]
C:\Users\Mateusz\Documents\Arduino\libraries\MPU6050/MPU6050_9Axis_MotionApps41.h:763:91: error: no 'uint8_t MPU6050::dmpGetLinearAccel(VectorInt16*, VectorInt16*, VectorFloat*)' member function declared in class 'MPU6050'
 uint8_t MPU6050::dmpGetLinearAccel(VectorInt16 *v, VectorInt16 *vRaw, VectorFloat *gravity) {
                                                                                           ^
C:\Users\Mateusz\Documents\Arduino\libraries\MPU6050/MPU6050_9Axis_MotionApps41.h:771:92: error: no 'uint8_t MPU6050::dmpGetLinearAccelInWorld(VectorInt16*, VectorInt16*, Quaternion*)' member function declared in class 'MPU6050'
 uint8_t MPU6050::dmpGetLinearAccelInWorld(VectorInt16 *v, VectorInt16 *vReal, Quaternion *q) {
                                                                                            ^
[... cut ...]
C:\Users\Mateusz\Documents\Arduino\libraries\MPU6050/MPU6050_9Axis_MotionApps41.h:823:84: error: no 'uint8_t MPU6050::dmpReadAndProcessFIFOPacket(uint8_t, uint8_t*)' member function declared in class 'MPU6050'
 uint8_t MPU6050::dmpReadAndProcessFIFOPacket(uint8_t numPackets, uint8_t *processed) {
                                                                                    ^
C:\Users\Mateusz\Documents\Arduino\libraries\MPU6050/MPU6050_9Axis_MotionApps41.h:848:40: error: no 'uint16_t MPU6050::dmpGetFIFOPacketSize()' member function declared in class 'MPU6050'
 uint16_t MPU6050::dmpGetFIFOPacketSize() {
                                        ^
Thank You in advance

jrowberg, 2017-05-22:
Hello @mateush888,

You seem to be compiling a project that isn't in the example set within the repository here. Would it be possible for you to provide the source file(s) that did not come from the repo so that we can look over them?

At the very least, it looks like you have included both `MPU6050_6Axis_MotionApps20.h` and `MPU6050_9Axis_MotionApps41.h` from your main .ino file, which will not work. These are mutually exclusive blocks of code as currently written, so you should only include on of them.

mateush888, 2017-05-23:
Thank You very much, @jrowberg ! All files came from Your repos on GitHub, I did not make any changes. I included only one library, as You mentioned earlier - Now everything works. The issue can be closed :)

mateush888, 2017-05-23:
I now figured out, that the code I used, wasnt exactly from Your repos, I am pasting it below. Really appreciate Your help, @jrowberg. Keep up the good work!

#include <math.h>
#include <Wire.h>
#include <helper_3dmath.h>
#include <MPU6050.h>
#include <I2Cdev.h>

///////////////////////////////////////////////////////////////////////////////////////
//THIS IS A DEMO SOFTWARE JUST FOR EXPERIMENT PURPOER IN A NONCOMERTIAL ACTIVITY
//Version: 1.0 (AUG, 2016)

//Gyro - Arduino UNO R3
//VCC  -  5V
//GND  -  GND
//SDA  -  A4
//SCL  -  A5
//INT - port-2

//Declaring some global variables
int gyro_x, gyro_y, gyro_z;
long gyro_x_cal, gyro_y_cal, gyro_z_cal;
boolean set_gyro_angles;

long acc_x, acc_y, acc_z, acc_total_vector;
float angle_roll_acc, angle_pitch_acc;

float angle_pitch, angle_roll;
int angle_pitch_buffer, angle_roll_buffer;
float angle_pitch_output, angle_roll_output;

long loop_timer;
int temp;

void setup() {
  Wire.begin();                                                        //Start I2C as master
  setup_mpu_6050_registers();                                          //Setup the registers of the MPU-6050 
  for (int cal_int = 0; cal_int < 1000 ; cal_int ++){                  //Read the raw acc and gyro data from the MPU-6050 for 1000 times
    read_mpu_6050_data();                                             
    gyro_x_cal += gyro_x;                                              //Add the gyro x offset to the gyro_x_cal variable
    gyro_y_cal += gyro_y;                                              //Add the gyro y offset to the gyro_y_cal variable
    gyro_z_cal += gyro_z;                                              //Add the gyro z offset to the gyro_z_cal variable
    delay(3);                                                          //Delay 3us to have 250Hz for-loop
  }

  // divide by 1000 to get avarage offset
  gyro_x_cal /= 1000;                                                 
  gyro_y_cal /= 1000;                                                 
  gyro_z_cal /= 1000;                                                 
  Serial.begin(115200);
  loop_timer = micros();                                               //Reset the loop timer
}

void loop(){

  read_mpu_6050_data();   
 //Subtract the offset values from the raw gyro values
  gyro_x -= gyro_x_cal;                                                
  gyro_y -= gyro_y_cal;                                                
  gyro_z -= gyro_z_cal;                                                
         
  //Gyro angle calculations . Note 0.0000611 = 1 / (250Hz x 65.5)
  angle_pitch += gyro_x * 0.0000611;                                   //Calculate the traveled pitch angle and add this to the angle_pitch variable
  angle_roll += gyro_y * 0.0000611;                                    //Calculate the traveled roll angle and add this to the angle_roll variable
  //0.000001066 = 0.0000611 * (3.142(PI) / 180degr) The Arduino sin function is in radians
  angle_pitch += angle_roll * sin(gyro_z * 0.000001066);               //If the IMU has yawed transfer the roll angle to the pitch angel
  angle_roll -= angle_pitch * sin(gyro_z * 0.000001066);               //If the IMU has yawed transfer the pitch angle to the roll angel
  
  //Accelerometer angle calculations
  acc_total_vector = sqrt((acc_x*acc_x)+(acc_y*acc_y)+(acc_z*acc_z));  //Calculate the total accelerometer vector
[... 34 lines cut ...]
  Wire.beginTransmission(0x68);                                        //Start communicating with the MPU-6050
  Wire.write(0x1C);                                                    //Send the requested starting register
  Wire.write(0x10);                                                    //Set the requested starting register
  Wire.endTransmission();                                             
  //Configure the gyro (500dps full scale)
  Wire.beginTransmission(0x68);                                        //Start communicating with the MPU-6050
  Wire.write(0x1B);                                                    //Send the requested starting register
  Wire.write(0x08);                                                    //Set the requested starting register
  Wire.endTransmission();                                             
}

void read_mpu_6050_data(){                                             //Subroutine for reading the raw gyro and accelerometer data
  Wire.beginTransmission(0x68);                                        //Start communicating with the MPU-6050
  Wire.write(0x3B);                                                    //Send the requested starting register
  Wire.endTransmission();                                             //End the transmission
  Wire.requestFrom(0x68,14);                                           //Request 14 bytes from the MPU-6050
  while(Wire.available() < 14);                                        //Wait until all the bytes are received
  acc_x = Wire.read()<<8|Wire.read();                                  
  acc_y = Wire.read()<<8|Wire.read();                                  
  acc_z = Wire.read()<<8|Wire.read();                                  
  temp = Wire.read()<<8|Wire.read();                                   
  gyro_x = Wire.read()<<8|Wire.read();                                 
  gyro_y = Wire.read()<<8|Wire.read();                                 
  gyro_z = Wire.read()<<8|Wire.read();                                 
}