jrowberg/i2cdevlib #311: Why these libraries are not in the arduino built-in library manager?

Bimblis, 2017-05-23:
Why these libraries are not in the arduino built-in library manager?

per1234, 2017-09-10:
For Library Manager compatibility it is required for the repository to contain a single library in the root of the repository rather than the current system of cramming a bunch of libraries for a variety of platforms into a slew of subfolders. This could be accomplished by merging the separate libraries together into a single Arduino library or breaking it into a repository for each library.

Details https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

jrowberg, 2017-09-24:
Better repo organization is something I'd very much like to address--not just for the Arduino IDE integration, but for improved maintainability in general, and to avoid forcing most users who are really just interested in one platform to download the whole thing.

I'd actually like to overhaul the whole thing with a fresh implementation of the same idea, now that we have years of experience as a community seeing where the library excels and where its (sometimes significant) limitations are. I've got a bunch of ideas written down that I think would significantly improve the compartmentalization and flexibility of the whole collection, and make it easier to contribute new devices and code features.

Now if only I can prioritize this effort over so many other things I have going on...

closed 2021-09-26 by jrowberg