jrowberg/i2cdevlib #315: i2cdevlib ² u8g2

getget1980, 2017-06-09:
Hello, I'm using MPU6050 Lib and u8g2 lib (Software SPI)

Arduino Mega 2560
Wiring : MPU 6050 on SDA Pin 20/SCL Pin 21/Interrupt Pin 2. Glcd Screen on 3,8,9.

MPU 6050 Sketch worls really fine on its own
u8g2 also works fine on its own.

When Adding screen processing in  while (!mpuInterrupt && fifoCount < packetSize) { } loop, screen does not work.

Adding screen processing at the end of the main loop makes screen and YPR work, BUT only for a random moment (between 0 and 300s).

Tried a lot of thing but can't find anything to make both of them working.

Any idea ?

Looks like MPU6050 lib gets stuck at  while (fifoCount < packetSize) fifoCount = mpu.getFIFOCount();

Any help would be appreciated :)

Thanks.

`#include <Arduino.h>
#include <SPI.h>
#include <U8g2lib.h>

#include "bitmaps.h"

int cycle;

/* ------------------ Includes Gyroscope --------------------------- */
// I2Cdev and MPU6050 must be installed as libraries, or else the .cpp/.h files
// for both classes must be in the include path of your project
#include "I2Cdev.h"

#include "MPU6050_6Axis_MotionApps20.h"
//#include "MPU6050.h" // not necessary if using MotionApps include file

// Arduino Wire library is required if I2Cdev I2CDEV_ARDUINO_WIRE implementation
// is used in I2Cdev.h
#if I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE
#include "Wire.h"
#endif

// class default I2C address is 0x68
// specific I2C addresses may be passed as a parameter here
// AD0 low = 0x68 (default for SparkFun breakout and InvenSense evaluation board)
// AD0 high = 0x69
MPU6050 mpu;
//MPU6050 mpu(0x69); // <-- use for AD0 high

// MPU control/status vars
bool dmpReady = false;  // set true if DMP init was successful
uint8_t mpuIntStatus;   // holds actual interrupt status byte from MPU
uint8_t devStatus;      // return status after each device operation (0 = success, !0 = error)
uint16_t packetSize;    // expected DMP packet size (default is 42 bytes)
uint16_t fifoCount;     // count of all bytes currently in FIFO
uint8_t fifoBuffer[64]; // FIFO storage buffer

// orientation/motion vars
Quaternion q;           // [w, x, y, z]         quaternion container
VectorInt16 aa;         // [x, y, z]            accel sensor measurements
VectorInt16 aaReal;     // [x, y, z]            gravity-free accel sensor measurements
VectorInt16 aaWorld;    // [x, y, z]            world-frame accel sensor measurements
VectorFloat gravity;    // [x, y, z]            gravity vector
float euler[3];         // [psi, theta, phi]    Euler angle container
float ypr[3];           // [yaw, pitch, roll]   yaw/pitch/roll container and gravity vector
float temp;

volatile bool mpuInterrupt = false;     // indicates whether MPU interrupt pin has gone high
[... 172 lines cut ...]
  }
  u8g2.clearBuffer();
  u8g2.setFont(u8g2_font_6x13_tf);
  u8g2.setCursor(0, 53);
  u8g2.print("Tilt:");
  u8g2.print(ypr[2] * 180 / M_PI, 0);
  u8g2.print("°");
  u8g2.setCursor(98, 0);
  u8g2.print("40.2v");
  u8g2.setCursor(0, 0);
  u8g2.print(temp, 1);
  u8g2.print("°C");
  u8g2.setCursor(64, 53);
  u8g2.print(millis() / 1000);
  u8g2.setFont(u8g2_font_logisoso32_tf);
  u8g2.setCursor(39, 15);
  u8g2.print("75%");
  u8g2.drawXBM(54, 0, 19, 9, battery);
  u8g2.sendBuffer();
  cycle = 0;
}

/*--------------------Fin Gestion affichages -------------------------------*/
}
`

danieldeesousa, 2017-10-25:
I am also having this problem. Did you fix it?

getget1980, 2017-10-25:
No, I used a second arduino to manage LCD Screen :(

danieldeesousa, 2017-10-25:
Problem solved.

Change U8G2_SSD1306_128X64_NONAME_2_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE); to U8G2_SSD1306_128X64_NONAME_2_HW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);