jrowberg/i2cdevlib #320: em_cmu.h not found

rbrody8, 2017-07-18:
I am trying to use the MPU6050 library. When i try to upload the IMU_Zero example though, I get a linker error in I2Cdev.h saying certain em device libraries are missing (i.e. em_cmu.h, em_device.h, em_bitband.h, etc.), and I can't seem to find them anywhere in the I2Cdevlib.

To get the libraries to work, I copied the .cpp/.h files from the .zip I downloaded into a subfolder of the directory my sketch is in, and I modified each #include statement accordingly.

Is there something I'm overlooking? Do you have a copy of the missing libraries I could use?

Thanks

jkleve, 2017-07-20:
I'm guessing those files come from a different library. [https://github.com/EnergyMicro/emlib](https://github.com/EnergyMicro/emlib)

jrowberg, 2017-07-20:
I'm guessing you dropped the entire repo into your Arduino Libraries folder. The header files noted in the error are only for the EFM32 port of the library. For your use case, you should put only the "I2Cdev" and "MPU6050" folders from the /Arduino subfolder on the repo inside your libraries folder

rbrody8, 2017-07-25:
Thanks! I put the wrong I2Cdev folder into my library folder...not too sure how that happened

closed 2017-07-25 by rbrody8

Dhairya-Patel-design, 2021-05-13:
I am trying to use the MPU6050 library. When I try to upload the MPU6050_raw example though, I get an error in saying certain em device libraries are missing (i.e. em_cmu.h, em_device.h, em_bitband.h, etc.), and I can't seem to find them anywhere in the I2Cdevlib.

Is there something I'm overlooking? Do you have a copy of the missing libraries I could use?

Thanks