jrowberg/i2cdevlib #325: How to acces mpu6050 gy 87 inclue Hmc5883l

jzishwa, 2017-09-05:
I want to use yaw data, with hmc5883l for stable data. Anyone can help me?

italogfernandes, 2017-09-06:
https://github.com/kriswiner/MPU6050/wiki/Affordable-9-DoF-Sensor-Fusion

jzishwa, 2017-09-06:
i can't find mpu6050+hmc5883l code, just mpu9250. can you help me to give its code, thank's

italogfernandes, 2017-09-06:
Neither the MPU6050_6Axis_MotionApps20.h or MPU6050_9Axis_MotionApps41.h work with the hmc5883l.
For get the  values from the hmc5883l you need to use this function in the setup:
`void setI2CBypassEnabled(bool enabled);`
Line 1507 of this file: https://github.com/jrowberg/i2cdevlib/blob/master/Arduino/MPU6050/MPU6050.cpp

Enabling this function allows you to use the hmc5883l library for get the mag readings.

To fuse the mag values to the quaternion, i did'nt find any easy way. My final approache was to do the sensor fusion using Madgwick Algorithm.