jrowberg/i2cdevlib #326: getMotion6() and dmpGetAccel return different values

ladislas, 2017-09-07:
modfying the MPU6050_DMP6.ino sketch to also output raw accel values with `getMotion6()`, I get the correct `16384` but get half the value from `dmpGetAccel()`, around `8180`.

I've researched the code but can't find the reason. Is that a normal behavior?

italogfernandes, 2017-09-07:
The getMotion6 read the accel values direct from the register 0x3B and the next 6. That is the pure reading of the accelerometer. 
The dmpGetAccel() should be reading the accel from the DMP fifo. That is, the accel values the the dmp uses to calculate quaternions and so on. By my readings, I think the accel in the dmp fifo is filtered by a Digital Low Pass Filter (DLPF), updated at 200Hz and others peculiar features.

So the Accel from getMotio6() has a LSB of 16384 ((2^16)/4 = 16384) (4 because it's from -2g  to 2g). 
And the Accel from the DMP FIFO (dmpGetAccel), has a LSB of 2^13 = 8192, i don't know why, but it's correct and should be because of the DMP proprietary algorithm.

ladislas, 2017-09-08:
Hmm, that's interesting.

Also the documentation in `MPU6050.h` says:

```
AFS_SEL | Full Scale Range | LSB Sensitivity
--------+------------------+----------------
0       | +/- 2g           | 8192 LSB/mg
1       | +/- 4g           | 4096 LSB/mg
2       | +/- 8g           | 2048 LSB/mg
3       | +/- 16g          | 1024 LSB/mg
```

So I'm not sure I understand anymore...

ladislas, 2017-09-08:
> And the Accel from the DMP FIFO (dmpGetAccel), has a LSB of 2^13 = 8192, i don't know why, but it's correct and should be because of the DMP proprietary algorithm.

@italogfernandes you say it's correct because everyone is having the same thing and it works with everything else? what values should I use then?

italogfernandes, 2017-09-08:
@ladislas I usually prefer the raw accel values, using ` getMotion6()` and 16384. And using another algorithm to do the sensor fusion. I think both options will give you good readings.

ladislas, 2017-09-08:
thanks a lot! :)