jrowberg/i2cdevlib #331: Compilation error for MKZero (SAMD21 based Arduino board)

Protonerd, 2017-10-03:
Hi,

I'm having an issue compiling a code using I2C communication with the MPU6050 accelro/gyrosensor. The code works fine with all AVR based micros. But has the following error when compiling for the MKZero (samd21, ARM M0+ based processor). Any idea why?

In file included from D:\Arduino\LS_CO\libraries\I2Cdev\I2Cdev.h:80:0,

                 from D:\Arduino\LS_CO\libraries\I2Cdev\I2Cdev.cpp:46:

D:\Arduino\LS_CO\libraries\I2Cdev\I2Cdev.cpp: In static member function 'static int8_t I2Cdev::readBytes(uint8_t, uint8_t, uint8_t, uint8_t*, uint16_t)':

D:\Arduino\LS_CO\libraries\I2Cdev\I2Cdev.cpp:276:62: error: 'BUFFER_LENGTH' was not declared in this scope

             for (uint8_t k = 0; k < length; k += min(length, BUFFER_LENGTH)) {

                                                              ^

C:\Users\MunkaGep\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.16\cores\arduino/Arduino.h:100:24: note: in definition of macro 'min'

 #define min(a,b) ((a)<(b)?(a):(b))

                        ^

D:\Arduino\LS_CO\libraries\I2Cdev\I2Cdev.cpp: In static member function 'static int8_t I2Cdev::readWords(uint8_t, uint8_t, uint8_t, uint16_t*, uint16_t)':

D:\Arduino\LS_CO\libraries\I2Cdev\I2Cdev.cpp:414:70: error: 'BUFFER_LENGTH' was not declared in this scope

             for (uint8_t k = 0; k < length * 2; k += min(length * 2, BUFFER_LENGTH)) {

                                                                      ^

C:\Users\MunkaGep\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.16\cores\arduino/Arduino.h:100:24: note: in definition of macro 'min'

 #define min(a,b) ((a)<(b)?(a):(b))

danieldeesousa, 2017-10-31:
I have the same error. Did you find a solution?

Protonerd, 2017-11-06:
Yes, I found a hint somewhere that only starting from a certain lib version SAMD21 boards are supported. I have my notes at home, but I think the info should be easy to find, I installed a new lib version and I did not have the issue any more.

jrowberg, 2021-09-25:
Errors related to undefined `BUFFER_LENGTH` have now been resolved with an addition to `I2Cdev.h` detecting and/or defining a suitable length for many platforms.

closed 2021-09-25 by jrowberg